horchi/linux-p4d #5: Segfault

ranseyer, 2016-04-01:
Hi,

im not shure if this issue is based in p4d...

a) maybe ist based in my serverconfig

b) if not, here it is:
kernel: [306212.508756] p4[28564]: segfault at 0 ip 0000000000405aa3 sp 00007ffdb284d0e0 error 6 in p4[400000+12000]

philipphunger, 2016-04-01:
waht system do you use?

ranseyer, 2016-04-01:
I use ubuntu 14.04 on an Intel Server. Kernel is 4.2. 
(is this the answer you wanted?)

philipphunger, 2016-04-01:
yes can you check your ram is ok?
wen you boot the server it wil say:
Ubuntu
Ubuntu Recovery
Memtest    <--- this is what you schould boot into to check if your ram is ok

ranseyer, 2016-04-01:
I can check it, but not now. The system is a little bit bigger an needs 5 Minutes for booting, and has to be online now.

horchi, 2016-04-05:
if it is still crashing please post a full backtrace with debug symbols

ranseyer, 2016-04-05:
Hi, 
RAM on the Server is OK. Last Segfault was 4 Days before.
> Apr  1 06:30:02 storage kernel: [289092.967201] p4[52223]: segfault at 0 ip 0000000000405aa3 sp 00007fff8e998b90 error 6 in p4[400000+12000]
(And i had a lot of segfaults)

At 28. of March a "USB Device" connected to the Server died, an the PCI-DVB devices are not usable...

So i looks that the error is gone. Sorry for aksing...

So i would be happy if could move all p4d log-entries from syslog to another file.

More OT: I hope you have seen this very good solution to connect a message broker (MQTT) to the P4D: https://forum.fhem.de/index.php/topic,30088.msg293487.html#msg293487
To this message broker you can connect FHEM an a lot of other aplications.
Thanks for the great work on P4D.

ranseyer, 2016-04-07:
Segfault again...

But the reason is maybe another software, Not P4D: p4tomqtt.pl

So i fear, i have to ask the expert for this scripts after some investogation... 

```
Apr  7 12:18:01 storage CRON[11700]: (root) CMD (/usr/bin/perl /usr/local/bin/p4tomqtt.pl >> /dev/null 2>&1)
Apr  7 12:18:02 storage kernel: [ 3210.060053] p4[11702]: segfault at 0 ip 0000000000405f45 sp 00007ffef4018730 error 6 in p4[400000+13000]
Apr  7 12:21:01 storage CRON[12149]: (root) CMD (/usr/bin/perl /usr/local/bin/p4tomqtt.pl >> /dev/null 2>&1)
Apr  7 12:21:02 storage kernel: [ 3390.650095] p4[12151]: segfault at 0 ip 0000000000405f45 sp 00007fffa66668f0 error 6 in p4[400000+13000]
Apr  7 12:24:01 storage CRON[12577]: (root) CMD (/usr/bin/perl /usr/local/bin/p4tomqtt.pl >> /dev/null 2>&1)
Apr  7 12:24:02 storage kernel: [ 3570.241071] p4[12579]: segfault at 0 ip 0000000000405f45 sp 00007fff53747fb0 error 6 in p4[400000+13000]
```

```
1460025003: New client connected from 192.168.1.2 as Net::MQTT::Simple[13429] (c1, k60).
1460025006: Socket error on client Net::MQTT::Simple[13429], disconnecting.
1460025182: New connection from 192.168.1.2 on port 1883.
1460025182: New client connected from 192.168.1.2 as Net::MQTT::Simple[13873] (c1, k60).
1460025185: Socket error on client Net::MQTT::Simple[13873], disconnecting.
```

closed 2017-01-12 by horchi