horchi/linux-p4d #8: p4d mysql-server password bug

CramericaIndustries, 2017-01-11:
Hi @horchi,
I've spent a few hours on figuring out why p4d wasn't able to connect to my mysql server, whilst I was able to connect to the mysql-database manually via the mysql-command-line-client using the same credentials.

Output was:
> root@51d9c6382bab:/tmp# /usr/local/bin/p4d -n -t
> Read 8 option from /etc/p4d/p4d.conf
> Info: Calling mysql_library_init()
> Dictionary '/etc/p4d/p4d.dat' loaded
> Try conneting to database
> Checking database connection ...
> Calling mysql_init(324)
> Closing mysql connection and calling mysql_thread_end(324)
> Error, connecting to database at 'mysql' on port (3306) failed
> Fatal: Initial database connect failed, aborting
> Initialization failed, see syslog for details
> Info: Released the last usage of mysql_lib, calling mysql_library_end() now

The cause of the problem was that my password contained a hash-sign ('#'). Then, as soon as I changed my password to a string consisting only of numbers and letters, p4d connected immediatelly.
Therefore I guess there must be a bug in the code that handles the mysql password.

horchi, 2017-01-12:
Hi, yes you are right!

I made to changes:
1.) the error message now clearer like:
SQL-Error in 'connecting to database' - Access denied for user 'p4'@'localhost' (using password: YES) (1045)
2.) login with passwords containing "#" is working now. It habe to be masked with \ since it was the comment sign of the conf file

CramericaIndustries, 2017-01-12:
Hey, that was fast! Thank you very much!

closed 2017-01-12 by horchi