horchi/linux-p4d #9: Spalten der Tabelle errors nicht konsistent spezifiert

Ronson1909, 2017-01-17:
Die time Spalte existiert noch in der Datenbank, in p4d.dat aber nicht mehr. Das führt dazu, dass eine Warnung kommt, die Spalte könne mit DROP gelöscht werden. Tut man das aber, funktioniert die error.php nicht mehr. Da gibt es offenbar in ORDER BY noch eine Referenz.
Wozu gibt es außerdem noch die Definitionen in tabledef.c wenn sie (vor allem?) in p4d.dat stecken?

horchi, 2017-01-18:
>> Die time Spalte existiert noch in der Datenbank, in p4d.dat aber nicht mehr. Das führt dazu, dass eine Warnung kommt, die Spalte könne mit DROP gelöscht werden,

ja ist richtig so, will nicht automatisch droppen.

>> Tut man das aber, funktioniert die error.php nicht mehr. Da gibt es offenbar in ORDER BY noch eine  Referenz.

das hatte ich auch umgestellt, hast du die error.php aktualisiert? Siehe: https://github.com/horchi/linux-p4d/blob/master/htdocs/error.php

>> Wozu gibt es außerdem noch die Definitionen in tabledef.c wenn sie (vor allem?) in p4d.dat stecken?
wird nicht mehr verwendet, ist Altlast, werde ich aus dem git nehmen

closed 2017-01-18 by horchi

Ronson1909, 2017-01-18:
Vielen Dank. Wird error.php nicht per make install-web aktualisiert? Wenn nicht, dann lag es daran.

horchi, 2017-01-18:
Doch make install-web installiert das mit. Ist es bei dir nicht im richtigen Pfad angekommen? 
Ggf. hast du dire php's unter /var/... doppelt, einmal von make install-web und einmal vom install Skipt von Phillip?

Ich habe ihm dazu schon per Mail einen Update seines Skriptes geschickt damit es das künftig auch per make macht, er hat aber noch nicht geantwortet.

Ronson1909, 2017-01-18:
Doppelt hab ich sie eigentlich nicht (wie auch?). Vielleicht habe auch ich einen Fehler gemacht und z.B. das sudo bei make install-web vergessen und die Fehler nicht wahrgenommen.

Danke. Und: Super Projekt!!!

horchi, 2017-01-18:
:) gerne!

Mit doppelt meine ich das die Pfade abweichen können wo make install und das install.sh hin installieren.
Aber läuft jetzt und die error.php passt?

Ronson1909, 2017-01-18:
Ja, läuft. Danke.

Ronson1909, 2017-01-18:
Da ich auch andere Aktualisierungen der php Dateien nicht gefunden habe, habe ich mir nochmal das make install-web angesehen. Dabei ist mir dann erst aufgefallen, dass die Variable WEBDEST aus Make.config genutzt wird und diese auf /var/lib/p4 gesetzt ist, statt auf /var/www/html
War ich da nur zu unwissend, dass ich auch die Make.config anpassen muss (dann wäre ein Hinweis in den Dokus gut), oder ist das ein Fehler? In /var/lib/p4 bringt es doch nichts, außer man linkt das nach /var/www/html oder?

horchi, 2017-01-19:
ich habe den Apache auf meinem Server nicht direkt auf dem Raspi, dort liegen alle Seiten die aus den Paketen kommen m.E. unter /var/lib/. In /etc/apache2/conf.d/ habe ich für jede Seite eine Konfiguration wie diese hier: https://github.com/horchi/linux-p4d/blob/master/apache2/p4.conf

Was der übliche Standard Pfad für Webseiten des Apache ist weiß ich auch nicht. 

Du musst die Make.config nicht unbedingt anpassen (geht ja mit jedem git update verloren). Man kann beim make Aufruf die Variablen auch überschreiben:
make install-web WEBDEST= /var/www/html

Nur, wenn das der Standard sein soll, wo würden dann weitere Seiten untergebracht?
Ist aber auch wurscht auf dem Raspi an der Heizung haben die meisten sicher nur die p4 Seite ;)