horchi/linux-p4d #10: Menus floating over each other and into content area

Ronson1909, 2017-01-26:
I'm using 0.2.24-GITa140b8b / 0.2.25 and current Chrome version.

Menus are floating over each other (e.g. in Setup) and into content area, such that text is partly not visible and buttons are also partly hidden.
[image]

horchi, 2017-01-26:
Can't reproduce using Chrome 56.0.2924.59 beta or Firefox 50.1.0

can you please try resatrting the browser, realoding the page by strg-shift-r (not only f5) and if this didn't help deleting the browser cache.

Ronson1909, 2017-01-26:
Yes, you're right. It was a caching issue.
However, on mobile devices (or in general on small-width devices or windows) the problem is similar because height is fixed and if buttons wrap this is not reflected. This is also true for second level menus (e.g. in setup).

[image]

Ronson1909, 2017-01-26:
As it doesn't seem to be possible to set position fixed to variable height menu with css only, this is a proposal for solution (inspired by http://stackoverflow.com/a/30489281/7475802):

Add jquery to header:

`echo "    <script type=\"text/JavaScript\" src=\"https://ajax.googleapis.com/ajax/libs/jquery/3.1.1/jquery.min.js\"></script>\n";
`

Remove height of menu (or set to auto) and set its top to 0
Add this javascript code:

`<script>$(document).ready(function() {
    var contentPlacement = $('.menu').position().top + $('.menu').height();
    $('.content').css('margin-top',contentPlacement);
});
`

Maybe it's better to use ids and # reference instead of .class reference in jquery code

horchi, 2017-01-27:
ja das Problem ist mir bekannt, hab mir schon einige Wege angesehen und hier auch schon was mit javascript gebaut. Sodass es am Browser fix ist und am Smartphone rasuscrollt da es dort zu viel Platz braucht. Dann aber beim zurück scrollen direkt eingeblendet wird damit man nicht ganz hoch muss.
Gefällt mir irgendwie alles noch nicht so recht bzw. klappt noch nicht so rund. Bin eher in C++ als in javascript / php / html zuhause ;)
Denke ggf könnte am Handy so ein 'Burger' Menü ganz gut sein?

Ronson1909, 2017-01-28:
Du meinst die drei waagrechten Linien und erst dann öffnet sich das Menü? Ja, sicher nicht schlecht. Deine genannte Variante "beim Hochscrollen einblenden, beim Runterscrollen verschwinden" ist aber vielleicht besser, weil ein Klick weniger für den User?
Bin leider auch kein js/php/html/css Guru, sodass ich da leider keine besonders gute Unterstützung leisten kann.

Mein Vorschlag oben, wäre aber zumindest ein Quickfix, der das Problem an sich erst Mal löst. Optimieren kann man ja danach noch weiter !?

horchi, 2020-11-01:
ist mir der neuen JS basierten Version erledigt

closed 2020-11-01 by horchi