horchi/linux-p4d #14: Can't build

SaimenSays, 2020-02-14:
What am I doing wrong?
On `make install` I got many errors. When editing db.h and changing `my_bool` to `bool` as suggestet, it seems to be fixed. But why does it build on your side?

`g++ -c -ggdb -O0 -fPIC -std=c++11 -Wno-psabi -Wreturn-type -Wall -Wno-parentheses -Wformat -pedantic -Wunused-variable -Wunused-label -Wunused-value -Wunused-function -Wno-long-long -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -I/usr/include/mysql -I/usr/include/libxml2 -D_GNU_SOURCE -DTARGET='"p4d"' -DLOG_PREFIX='""' -DUSEUUID -DUSEMD5 -DUSELIBXML -DUSELIBARCHIVE -DUSEJSON -DUSEGUNZIP -D_GNU_SOURCE -DTARGET='"p4d"' -DDEAMON=P4d -DUSEMD5 -DGIT_REV='"932874c"' -o lib/db.o lib/db.c
In file included from lib/db.c:13:
lib/db.h:419:7: error: ‘my_bool’ does not name a type; did you mean ‘bool’?
       my_bool* getNullRef()                { return &nullValue; }
       ^~~~~~~
       bool
lib/db.h:431:7: error: ‘my_bool’ does not name a type; did you mean ‘bool’?
       my_bool nullValue;
       ^~~~~~~
       bool
lib/db.h: In member function ‘void cDbValue::clear()’:
lib/db.h:90:10: error: ‘nullValue’ was not declared in this scope
          nullValue = 1;
          ^~~~~~~~~
lib/db.h:90:10: note: suggested alternative: ‘numValue’
          nullValue = 1;
          ^~~~~~~~~
          numValue
lib/db.h: In member function ‘void cDbValue::setNull()’:
lib/db.h:115:18: error: ‘nullValue’ was not declared in this scope
          int n = nullValue;
                  ^~~~~~~~~
lib/db.h:115:18: note: suggested alternative: ‘numValue’
          int n = nullValue;
                  ^~~~~~~~~
                  numValue
lib/db.h: In member function ‘void cDbValue::setValue(const char*, int)’:
lib/db.h:175:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:175:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h:190:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:190:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h: In member function ‘void cDbValue::setValue(long int)’:
lib/db.h:218:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:218:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h:237:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:237:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h: In member function ‘void cDbValue::setValue(double)’:
lib/db.h:256:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:256:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h:264:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:264:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h:272:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:272:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h: In member function ‘void cDbValue::setBigintValue(int64_t)’:
lib/db.h:288:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:288:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h:297:13: error: ‘nullValue’ was not declared in this scope
             nullValue = 0;
             ^~~~~~~~~
lib/db.h:297:13: note: suggested alternative: ‘numValue’
             nullValue = 0;
             ^~~~~~~~~
             numValue
lib/db.h: In member function ‘int cDbValue::isNull()’:
lib/db.h:393:53: error: ‘nullValue’ was not declared in this scope
       int isNull()                         { return nullValue; }
                                                     ^~~~~~~~~
lib/db.h:393:53: note: suggested alternative: ‘numValue’
       int isNull()                         { return nullValue; }
                                                     ^~~~~~~~~
                                                     numValue
lib/db.c: In member function ‘int cDbStatement::appendBinding(cDbValue*, cDbService::BindType)’:
lib/db.c:495:36: error: ‘class cDbValue’ has no member named ‘getNullRef’; did you mean ‘setNull’?
       newBinding->is_null = value->getNullRef();
                                    ^~~~~~~~~~
                                    setNull
lib/db.c:505:36: error: ‘class cDbValue’ has no member named ‘getNullRef’; did you mean ‘setNull’?
       newBinding->is_null = value->getNullRef();
                                    ^~~~~~~~~~
                                    setNull
lib/db.c:514:37: error: ‘class cDbValue’ has no member named ‘getNullRef’; did you mean ‘setNull’?
       newBinding->is_null =  value->getNullRef();
                                     ^~~~~~~~~~
                                     setNull
lib/db.c:523:37: error: ‘class cDbValue’ has no member named ‘getNullRef’; did you mean ‘setNull’?
       newBinding->is_null =  value->getNullRef();
                                     ^~~~~~~~~~
                                     setNull
lib/db.c:533:37: error: ‘class cDbValue’ has no member named ‘getNullRef’; did you mean ‘setNull’?
       newBinding->is_null =  value->getNullRef();
                                     ^~~~~~~~~~
                                     setNull
lib/db.c:543:37: error: ‘class cDbValue’ has no member named ‘getNullRef’; did you mean ‘setNull’?
       newBinding->is_null =  value->getNullRef();
                                     ^~~~~~~~~~
                                     setNull
make: *** [Make.config:60: lib/db.o] Fehler 1
`

horchi, 2020-02-15:
the first error is your not readable formatted post.
Regarding the errors, The mysql header seems to be missing

horchi, 2020-03-23:
check the README about the dependencies - all required packeges should be listed there.
Pleas report is one missing

closed 2020-03-23 by horchi