horchi/linux-p4d #15: no install in debian buster mariadb

stixif, 2020-03-22:
nach 
make clean all HASSMQTT=yes

kommt diese Fehlermeldung

OSTYPE is Linux
make: mysql_config: Command not found
g++ -c -ggdb -O0 -pthread -fPIC -std=c++11 -Wno-psabi -Wreturn-type -Wall -Wno-p         arentheses -Wformat -pedantic -Wunused-variable -Wunused-label -Wunused-value -W         unused-function -Wno-long-long -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64  -I/us         r/include/libxml2 -D_GNU_SOURCE -DTARGET='"p4d"' -DLOG_PREFIX='""' -DUSEUUID -DU         SEMD5 -DUSELIBXML -DUSELIBARCHIVE -DUSEJSON -DUSEGUNZIP -D_GNU_SOURCE -DTARGET='         "p4d"' -DDEAMON=P4d -DUSEMD5 -DMQTT_HASS -DGIT_REV='"a6823de"' -o lib/db.o lib/d         b.c
lib/db.c:9:10: fatal error: errmsg.h: No such file or directory
 #include <errmsg.h>
          ^~~~~~~~~~
compilation terminated.
make: *** [Make.config:63: lib/db.o] Error 1

horchi, 2020-03-22:
hast du die im Readme beschrieben Abhängigkeiten erfüllt, also alle aufgelisteten SW Pakete installiert?

horchi, 2020-03-23:
README nochmal aktualisiert und Option zur Installation mittels Debian Package (für Raspbian Buster) hinzugefügt

closed 2020-03-23 by horchi