horchi/linux-p4d #16: Makefile: Ability to just build the `p4` binary.

dominic-miglar, 2020-10-12:
Hey there 👋

First of all, I want to thank you for this great project @horchi !

I am just using the `p4` binary of that project in my [froeling-reader-service](https://github.com/logreposit/froeling-reader-service) and have to build a lot of dependencies (e.g. libwebsockets, and so on and so forth..). It would be very useful if there has been a Makefile-target which just builds that particular binary.

horchi, 2020-11-01:
have you tried calling `make p4`, this shouldn't need the libwebsockets an the systemd.
Actually I have only systems which full fill a dependencies whereby I can't test it.
Please let me now if there are other needless dependencies.

horchi, 2022-05-03:
seems to have settled, I close the issue times

closed 2022-05-03 by horchi