horchi/linux-p4d #20: Befehl zum Abschalten des Kessels für Wartung

TazerReloaded, 2021-01-16:
Hallo zusammen,

ich habe parallel zu der Entstehung dieser Software meine eigene Lösung zum Ansteuern eines P4 mit S3200 entwickelt. Der Funktionsumfang ist recht ähnlich glaube ich: https://heizung.pc-id.de/
Was mir aktuell allerdings fehlt ist eine Möglichkeit, die Steuerung in den Zustand "Kessel aus" zu bekommen, also wie wenn man den Power-Knopf am Display drückt. Ich stelle momentan den Einschaltpunkt auf 10°C, dann lässt er den Puffer auskühlen, das scheint mir aber ein ziemlicher Hack zu sein, und funktioniert natürlich nicht wenn er schon brennt. Gibt es da einen passenden Befehl?

Wenn Interesse besteht, kann ich auch gerne Einsicht in meinen Code gewähren. Allerdings sind Teile davon bereits 10 Jahre alt, ich habe immer nur die Teile angefasst, die Probleme gemacht haben, bräuchte also zunächst etwas Pflege (C# mit .NET 2.0), zeichnet aber seit 2009 Daten auf.

Gruß
Peter

MiGeppert, 2021-01-17:
Hallo allerseits

Auch mich würde interessieren, wie man den Kessel ausschalten bzw. einschalten kann.
Bei mir steht ein "Dual" im Keller und ich mag eigentlich ausschließlich mit Stückholz feuern.
Ab und zu gibt's die Notwendigkeit, auch mal mit Pellets zu feuern, und das würde ich gerne von unterwegs aktivieren können.

Und nein, Fröling Connect mag ich nicht verwenden, obwohl ich ehrlich gesagt schon einige Male mit dem Gedanken gespielt habe.

Gruß
MiGeppert

horchi, 2022-05-03:
ich kennen keinen Weg das über die Schnittstelle zu steuern. Wenn Ihr einen findet macht einen neuen Issue auf und lasst es mich wissen. Auch mir fehlt diese Funktion.
Grüße Jörg

closed 2022-05-03 by horchi