horchi/linux-p4d #22: Mqtt not updated

SaimenSays, 2021-07-24:
First of all, thanks for this great piece of Software!

Yesterday I have updated from a years old version to actual from repo. After some struggeling, to get anything all dependencies solved and updated, I got it compiled and now working again on webinterface.
The I did the setup and entered MQTT server. It seems to work from mqtt side, because I see some topics once in a while at my server. Also in p4d.log only shows rare occurance of mqtt output. Even after a whole day not all values are transmitted once.
Can someone help me, what and where to check what is going wrong?

hoppel118, 2021-07-24:
Welche Version hat du kompiliert?

Ich hatte neulich das gleiche Problem. @horchi hatte dann was korrigiert und ein neues Paket für den Pi veröffentlicht.

Das war damals folgende Version: 0.7.8

Die läuft auch bei mir immer noch.

Gruß Hoppel

SaimenSays, 2021-07-24:
@hoppel118: Ich habe einen Pull des Repos gemacht und es selbst kompiliert, da ich die alte Version auch aus den Sources kompiliert hatte. In den letzten Monaten hat sich nichts getan, sollte es der aktuelle Master branch sein.
Ich denke inzwischen, dass es eventuell mit Überbleibsel der vorherigen Installation verklemmt hat.
Wenn es keine Vorschläge gibt, werde ich den PI wohl nochmal komplett neu aufsetzen müssen.

hoppel118, 2021-07-25:
Ach du hast einen Pi. Dann führe doch einfach das Installationsscript aus. ;)

SaimenSays, 2021-07-25:
Habe jetzt die SD-Karte komplett neu aufgesetzt, nun mit dem Install-Script. Jetzt läuft es zum Glück wieder, also hatte sich irgenwo etwas verklemmt.
Da ich die Messwerte sowieso per MQTT noch auf einem separaten Server speichere, kann ich auf die Historie in der MySQL DB verzichten.
Notiz am Rande: Leider wurden zwischenzeitlich alle Umlaute aus "Fühler", "Stellgröße", etc. entfernt. Somit musste ich nun vieles manuell in meiner Datenbank umkopieren :-/

Funktioniert das Script auch wiederholt, für ein Update?

hoppel118, 2021-07-25:
Umlaute in Sensorbezeichnungen sind nicht konform. Ich glaube, wir hatten die Diskussion damals bei der MQTT-Anpassung für FHEM und Openhab…

Jo, einfach das Script ausführen, um das Update durchzuführen.

Gruß

SaimenSays, 2021-07-26:
Ich hatte anfangs auch überlegt ob Englisch oder Deutsch;) Weiß nicht, ob es die Fröhling überhaupt außerhalb des deutschsprachigen Raum gibt...

closed 2021-07-26 by SaimenSays