horchi/linux-p4d #25: Visual Problems in new Web Interface

N3m3515, 2021-12-04:
Hello,

there are Problems with the new Dashboard UI:

[image]

there is no text on this position:
[image]

and the List is not scrollable anymore.

There was a collum named Beispiel before, i think it was really helpfull. Can we get it back?
[image]

Thanks for your great work
:-)

N3m3515, 2021-12-04:
And if i change my widgets to value, i cant click on them to open a chart:

[image]

horchi, 2021-12-04:
1) shorten The Name of the title or expand the widget width to factor 1.5 or 2
2) the list will be removed or replaced soon
3) must think about it, required a lot of code
4) I will add in one of the next versions

N3m3515, 2021-12-04:
> 1. shorten The Name of the title or expand the widget width to factor 1.5 or 2
> 2. the list will be removed or replaced soon
> 3. must think about it, required a lot of code
> 4. I will add in one of the next versions

Thank you for that.
I found another Issue:
Event though i changed the Images it does`nt seem to change it in the Dashboard:
[image]
 Can we also use the images in the state folder?
https://github.com/horchi/linux-p4d/tree/master/htdocs/img/state

horchi, 2021-12-04:
> Event though i changed the Images it does`nt seem to change it in the Dashboard:
I can't reproduce, here it changes soon as I click at preview

> Can we also use the images in the state folder?
okay I will add it

N3m3515, 2021-12-04:
> > Event though i changed the Images it does`nt seem to change it in the Dashboard:
> > I can't reproduce, here it changes soon as I click at preview

Thats odd, it seems that nothing i am doing is working here. No Browser changing or reloading will give me the other icons.

horchi, 2021-12-04:
You tried a full pare reload with ctrl-shift-r ?
Any errors at the browser console?

N3m3515, 2021-12-04:
> You tried a full pare reload with ctrl-shift-r ? Any errors at the browser console?

Yes did that. No errors in the concole or in syslog.
I have noticed, that there are two folders for the icons:

/var/lib/p4d/img/icon
/var/lib/p4/img/icon

[image]
[image]

the user rights are different and it seems that the new version is using /var/lib/p4d/

horchi, 2021-12-04:
yes, /var/lib/p4/ can be deleted

horchi, 2021-12-04:
you have noticed the update hint to 0.8.x here https://github.com/horchi/linux-p4d/blob/master/porting-to-0.8.0

N3m3515, 2021-12-04:
Yes i did. but didnt want to delete before everything is working. I just noticed the different user rights

horchi, 2021-12-04:
ah okay!
the rights don't matter until p4d is running as root

horchi, 2021-12-04:
okay now I can reproduce the problem with the images !

N3m3515, 2021-12-04:
> okay now I can reproduce the problem with the images !

great or not great :-)
i hope you can find the fix.

horchi, 2021-12-04:
Fixed - version 0.8.4

from HISTORY.h:
```
2021-12-04:  version 0.8.4
  - added:  All images below img to selection for icon widgets
  - change: Speed up dashboard display
  - change: Added chart view to value widgets
  - bugfix: Fixed problem with image selection for icon widgets

```

N3m3515, 2021-12-04:
ok i am compiling right now

N3m3515, 2021-12-04:
sadly the dashboard is now not working anymore:

[image]

horchi, 2021-12-04:
i have an idea, i redesigned the config item for the dashboard, and this happens at startup when it is still empty.
I now added an nearly empty empty config for the initial dashboard (version 0.8.5)

N3m3515, 2021-12-04:
ok trying again

horchi, 2021-12-04:
the reason for the change regarding the dashboard setup from:
`| 2021-12-04 18:02:25  | addrsDashboard | UD:0x03,UD:0x01,VA:0x76,VA:0x00,W1:0x5679341,VA:0x78,VA:0x15,VA:0x19,VA:0x71,DO:0x00,VA:0xca,VA:0xc9,SD:0x03,SC:0x05,SP:0x3e6,SP:0x3e7,SC:0x04 |
`
to:
| 2021-12-04 20:18:15  | dashboards     | {"dashboard":{"UD:0x03":true,"UD:0x01":true,"VA:0x76":true,"VA:0x00":true,"W1:0x5679341":true,"VA:0x78":true,"VA:0x15":true,"VA:0x19":true,"VA:0x71":true,"DO:0x00":true,"VA:0xc9":true,"VA:0xca":true,"SD:0x03":true,"SC:0x05":true}} |

is the preparation for multiple dashboard views

N3m3515, 2021-12-04:
i am still getting the same error. and now it is crashing constantly and restarting itself without any good error in syslog:

`Dec  4 20:32:19 heizung w1mqtt: File '/sys/bus/w1/devices/w1_bus_master1/w1_master_slaves' not found'
Dec  4 20:32:19 heizung w1mqtt: MQTT: Connecting publisher to 'tcp://localhost:1883' succeeded
Dec  4 20:32:19 heizung w1mqtt: Error: Writing '{"ping"}: true' (14) to topic 'p4d2mqtt/w1/ping' failed, result was 2147483665 'MQTT_ERROR_SOCKET_ERROR'
Dec  4 20:32:19 heizung w1mqtt: Info: Disconnecting ''
Dec  4 20:32:19 heizung w1mqtt: Error: mqtt_sync for connection '' failed, result was -2147483631 'MQTT_ERROR_SOCKET_ERROR'
Dec  4 20:32:22 heizung systemd[1]: p4d.service: Main process exited, code=killed, status=6/ABRT
Dec  4 20:32:22 heizung systemd[1]: p4d.service: Failed with result 'signal'.
Dec  4 20:32:29 heizung w1mqtt:  ... done
Dec  4 20:32:29 heizung w1mqtt: Updating ...
Dec  4 20:32:29 heizung w1mqtt: File '/sys/bus/w1/devices/w1_bus_master1/w1_master_slaves' not found'
Dec  4 20:32:29 heizung w1mqtt: MQTT: Connecting publisher to 'tcp://localhost:1883' succeeded
Dec  4 20:32:29 heizung w1mqtt: Error: mqtt_sync for connection '' failed, result was -2147483631 'MQTT_ERROR_SOCKET_ERROR'
Dec  4 20:32:29 heizung w1mqtt: Error: Writing '{"ping"}: true' (14) to topic 'p4d2mqtt/w1/ping' failed, result was 2147483665 'MQTT_ERROR_SOCKET_ERROR'
Dec  4 20:32:29 heizung w1mqtt: Info: Disconnecting ''
Dec  4 20:32:39 heizung w1mqtt:  ... done
Dec  4 20:32:39 heizung w1mqtt: Updating ...
Dec  4 20:32:39 heizung w1mqtt: File '/sys/bus/w1/devices/w1_bus_master1/w1_master_slaves' not found'
Dec  4 20:32:39 heizung w1mqtt: MQTT: Connecting publisher to 'tcp://localhost:1883' succeeded
Dec  4 20:32:39 heizung w1mqtt: Error: mqtt_sync for connection '' failed, result was -2147483631 'MQTT_ERROR_SOCKET_ERROR'
Dec  4 20:32:39 heizung w1mqtt: Error: Writing '{"ping"}: true' (14) to topic 'p4d2mqtt/w1/ping' failed, result was 2147483665 'MQTT_ERROR_SOCKET_ERROR'
Dec  4 20:32:39 heizung w1mqtt: Info: Disconnecting ''
Dec  4 20:32:49 heizung w1mqtt:  ... done
Dec  4 20:32:49 heizung w1mqtt: Updating ...
Dec  4 20:32:49 heizung w1mqtt: File '/sys/bus/w1/devices/w1_bus_master1/w1_master_slaves' not found'
Dec  4 20:32:49 heizung w1mqtt: MQTT: Connecting publisher to 'tcp://localhost:1883' succeeded
Dec  4 20:32:49 heizung w1mqtt: Error: mqtt_sync for connection '' failed, result was -2147483631 'MQTT_ERROR_SOCKET_ERROR'
Dec  4 20:32:49 heizung w1mqtt: Error: Writing '{"ping"}: true' (14) to topic 'p4d2mqtt/w1/ping' failed, result was 2147483665 'MQTT_ERROR_SOCKET_ERROR'
Dec  4 20:32:49 heizung w1mqtt: Info: Disconnecting ''
Dec  4 20:32:52 heizung systemd[1]: p4d.service: Service RestartSec=30s expired, scheduling restart.
Dec  4 20:32:52 heizung systemd[1]: p4d.service: Scheduled restart job, restart counter is at 3.
Dec  4 20:32:52 heizung systemd[1]: Stopped P4d Control Deamon.
Dec  4 20:32:52 heizung systemd[1]: Starting P4d Control Deamon...
Dec  4 20:32:52 heizung systemd[1]: Started P4d Control Deamon.
Dec  4 20:32:56 heizung sSMTP[703]: Creating SSL connection to host
Dec  4 20:32:56 heizung sSMTP[703]: SSL connection using ECDHE_RSA_AES_256_GCM_SHA384
Dec  4 20:32:57 heizung sSMTP[703]: Sent mail for schu_bernd@web.de (221 web.de Service closing transmission channel) uid=0 username=root outbytes=2487
`

horchi, 2021-12-04:
are there may be any helpful in /var/log/p4d.log ?
Can you poste me a backtrace

N3m3515, 2021-12-04:
I cant see anything special in the p4d.log:

`Dec  4 20:35:55 heizung p4d: Read 5 option from /etc/p4d/daemon.conf
Dec  4 20:35:55 heizung p4d: Info: Calling mysql_library_init()
Dec  4 20:35:55 heizung p4d: current locale is de_DE.UTF-8
Dec  4 20:35:55 heizung p4d: Detected UTF-8
Dec  4 20:35:55 heizung p4d: Dictionary '/etc/p4d/database.dat' loaded
Dec  4 20:35:55 heizung p4d: Try conneting to database
Dec  4 20:35:55 heizung p4d: Checking database connection ...
Dec  4 20:35:55 heizung p4d: Calling mysql_init(736)
Dec  4 20:35:55 heizung p4d: SQL client character now 'utf8'
Dec  4 20:35:55 heizung p4d: Checking table structure and indices ...
Dec  4 20:35:55 heizung p4d: Checking table 'config'
Dec  4 20:35:55 heizung p4d: Checking table 'errors'
Dec  4 20:35:55 heizung p4d: Checking table 'groups'
Dec  4 20:35:55 heizung p4d: Checking table 'menu'
Dec  4 20:35:55 heizung p4d: Checking table 'peaks'
Dec  4 20:35:55 heizung p4d: Checking table 'pellets'
Dec  4 20:35:55 heizung p4d: Checking table 'samples'
Dec  4 20:35:55 heizung p4d: Checking table 'schemaconf'
Dec  4 20:35:55 heizung p4d: Checking table 'scripts'
Dec  4 20:35:55 heizung p4d: Info: Field 'name' not used anymore, to remove it call 'ALTER TABLE scripts DROP COLUMN name;' manually
Dec  4 20:35:55 heizung p4d: Info: Field 'visible' not used anymore, to remove it call 'ALTER TABLE scripts DROP COLUMN visible;' manually
Dec  4 20:35:55 heizung p4d: Checking table 'sensoralert'
Dec  4 20:35:55 heizung p4d: Checking table 'timeranges'
Dec  4 20:35:55 heizung p4d: Checking table 'users'
Dec  4 20:35:55 heizung p4d: Checking table 'valuefacts'
Dec  4 20:35:55 heizung p4d: Info: Field 'maxscale' not used anymore, to remove it call 'ALTER TABLE valuefacts DROP COLUMN maxscale;' manually
Dec  4 20:35:55 heizung p4d: Info: Field 'ord' not used anymore, to remove it call 'ALTER TABLE valuefacts DROP COLUMN ord;' manually
Dec  4 20:35:55 heizung p4d: Info: Field 'value' not used anymore, to remove it call 'ALTER TABLE valuefacts DROP COLUMN value;' manually
Dec  4 20:35:55 heizung p4d: Closing mysql connection and calling mysql_thread_end(736)
Dec  4 20:35:55 heizung p4d: Checking table structure and indices succeeded
Dec  4 20:35:55 heizung p4d: Calling mysql_init(736)
Dec  4 20:35:55 heizung p4d: Connection to database established
Dec  4 20:35:55 heizung p4d: Info: LogLevel set to 1
Dec  4 20:35:55 heizung p4d: Loaded (3) states [:3:5:6]
Dec  4 20:35:55 heizung p4d: WebSocket Listener at port (80) established
Dec  4 20:35:55 heizung p4d: using libwebsocket version '4.3.99-v4.3.0-106-ge1a73c42'
Dec  4 20:35:56 heizung p4d: Init script value of 'SC:1' to 0,00
Dec  4 20:35:56 heizung p4d: Found script '/etc/p4d/scripts.d/example.sh' addr (1), unit '(null)'; result was [{ "kind":"status","value":0 }]
Dec  4 20:35:56 heizung p4d: Initialize python script '/etc/p4d/scripts.d/sensorExample.py'
Dec  4 20:35:56 heizung p4d: Python: PySensor::getData: VA:1  [{"address": 1, "type": "VA", "name": "Abgastemperatur_0x1", "title": "Abgastemperatur", "unit": "°C", "rights": 0, "peak": 214.0, "peakmin": 40.0, "value": 0.0, "last": 0, "valid": 0}]
Dec  4 20:35:56 heizung p4d: Init script value of 'SC:2' to 0,00
Dec  4 20:35:56 heizung p4d: Found script '/etc/p4d/scripts.d/sensorExample.py' addr (2), unit '°C'; result was [{"kind": "value", "title": "Python Test", "unit": "\u00b0C", "value": 0.0}]
Dec  4 20:35:56 heizung p4d: Loaded (3) states [:3:5:6]
Dec  4 20:35:56 heizung p4d: NO aggregateHistory configured!
Dec  4 20:35:56 heizung p4d: State changed to 'Heizen'
Dec  4 20:35:56 heizung p4d: ----- message '{"event": "daemonstate", "object": {"state": 0, "version": "0.8.5-GIT5895c9d", "runningsince": "1 Sekunde", "average0": 0.72, "average1": 0.59, "average2": 0.34}}'
Dec  4 20:35:56 heizung p4d: Updated 15 samples
Dec  4 20:35:56 heizung p4d: Updating error list
Dec  4 20:35:57 heizung p4d: Client '172.16.30.67' connected (0x72c00a20), ping time set to (60)
Dec  4 20:35:57 heizung p4d: Client login '172.16.30.67' (0x72c00a20) [{"event":"login","object":{"type":"active","user":"admin","page":"dashboard","token":"4A828388-8DA2-4C39-AAED-70AAB3623D1E_03.12.2021 18:53:38_admin"}}]
Dec  4 20:35:58 heizung p4d: Updating error list done in 2077ms
Dec  4 20:36:00 heizung p4d: Send mail 'Heizung - Status: Heizen' to 'bschu1985@gmail.com'
Dec  4 20:36:00 heizung p4d: ----- message '{"event": "s3200-state", "object": {"time": 1638646555, "state": 3, "stateinfo": "Heizen", "mode": 1, "modeinfo": "Brauchwasser", "version": "50.04.05.18", "image": "img/state/light/state-fire.gif"}}'
Dec  4 20:36:00 heizung p4d: Update script sensors
Dec  4 20:36:00 heizung p4d: Login of client 0x72c00a20; user 'admin'; type is 0; page dashboard
Dec  4 20:36:00 heizung p4d: ----- message '{"event": "config", "object": {"webUrl": "http://172.16.10.119:80", "webSsl": "0", "haUrl": "undefined", "heatingType": "S4turbo", "style": "dark", "iconSet": "light", "schema": "p4-2hk-puffer4", "dashboards": "", "chartRange": "1,50", "chartSensors": "VA:0x0,VA:0x1,VA:0x5d,VA:0x76,VA:0x77,VA:0x78,VA:0x1bf,", "interval": "60", "webPort": "80", "stateCheckInterval": "10", "arduinoInterval": "10", "ttyDevice": "/dev/ttyUSB0", "loglevel": "1", "mqttUrl": "", "tsync": "1", "maxTimeLeak": "10", "aggregateHistory": "0", "aggregateInterval": "5", "peakResetAt": "", "consumptionPerHour": "0", "mqttHassUrl": "", "mqttHassUser": "", "mqttHassPassword": "", "mqttDataTopic": "p4d2mqtt/sensor/<NAME>/state", "mqttHave [... cut ...] peakmin": 30.5, "value": 54.5}, "VA:0xc4": {"address": 196, "type": "VA", "name": "TemperaturdesZweitkessel_0xc4", "title": "Öl-Kessel", "unit": "°C", "rights": 0, "peak": 51.5, "peakmin": 31.5, "value": 42.5}}}'
Dec  4 20:42:33 heizung p4d: ----- message '{"event": "s3200-state", "object": {"time": 1638646945, "state": 3, "stateinfo": "Heizen", "mode": 1, "modeinfo": "Brauchwasser", "version": "50.04.05.18", "image": "img/state/light/state-fire.gif"}}'
`

How can i provide you with a backtrace?

horchi, 2021-12-04:
ich glaube ich habe es gefunden - bin dran (you seaking german, or?)

horchi, 2021-12-04:
okay should be fixed now. Sorry for the circumstances

N3m3515, 2021-12-04:
Ja Deutsch passt :-D

ich probiere aus

N3m3515, 2021-12-04:
ja die crashes scheinen weg zu sein aber das dashboard wird noch nicht richtig initialisiert:

[image]

horchi, 2021-12-04:
Kannst du das Ergebnis von

select * from config where name = ‚dashboards‘;

posten bitte

N3m3515, 2021-12-04:
Das wird das problem sein:

MySQL meldet: Dokumentation 
#1054 - Unbekanntes Tabellenfeld '‚dashboards‘' in where clause

horchi, 2021-12-04:
das dashbords muss in einfachen hochkomma stehen. Die sehen hier komisch aus ‚‘ der unten ist falsch.
Aber ich hab gesehen was ich wollte.

denke du musst nur dir row bei der der name = dashboards ist löschen 

`delete from config where name = ‚dashboards‘`

horchi, 2021-12-04:
aber mach die hochkomma manuell nicht copy paste

N3m3515, 2021-12-04:
Ja das wars. Jetzt gehts:

[image]

horchi, 2021-12-04:
Prima !

N3m3515, 2021-12-04:
Da muss ja dann was beim initialisieren des wertes in der database schief gelaufen sein? oder lag das nur an meiner updaterei?
Danke für deine Hilfe.

horchi, 2021-12-04:
Ich hatte den in der ersten Version nach meinem Umbau heute leer initialisiert, mein Fix danach hat dann bei dir nicht gefruchtet da er zu dem Zeitpunkt schon leer in der config Tabelle stand. Jeder der erst jetzt auf die Version von heute umstellt hat das Problem nicht

horchi, 2021-12-04:
Btw, die Reihenfolge der widgets kannst du mit drag & drop anpassen- hast du vermutlich schon bemerkt

N3m3515, 2021-12-04:
Sehr gut.
Ich habe jetzt die letzten 2 tage auch zusammengetragen wie ich das system aus einem frischen Raspbian Buster (heisst jetzt Legacy OS Lite) installiert habe:

cd /usr/src/ && rm -r libwebsockets && apt-get update && apt-get upgrade && apt-get install git build-essential pkg-config libssl-dev libjansson-dev libxml2-dev libcurl4-openssl-dev libssl-dev libmariadbclient-dev libmariadb-dev-compat uuid-dev cmake libcap-dev ssmtp mailutils haveged python3-dev wiringpi && git clone https://libwebsockets.org/repo/libwebsockets libwebsockets && cd libwebsockets && mkdir build && cd build && cmake .. && make -s && make -s install && cd /usr/src/ && rm -r linux-p4d/ && git clone https://github.com/horchi/linux-p4d/ && cd linux-p4d/ && make clean all && make install && reboot now

Kann man ja als Referenz für die Anleitung nutzen.

horchi, 2021-12-05:
Kennst du den Thread zum p4d im Holzheizer Forum? 
Für Buster auf dem Raspi hab ich ein fertiges deb

N3m3515, 2021-12-05:
Ja das habe ich gesehen. Leider hat das bei mir nicht funktioniert. Aber vllt wäre es eine gute Idee sich mal anzuschauen ob man das Projekt nicht in ein Docker image packen könnte.

horchi, 2021-12-05:
Wo hat es denn geklemmt?

N3m3515, 2021-12-05:
Der Service ist beim start gecrasht. Müsste es nochmal nachstellen um da jetzt genau sagen zu können worans lag.

horchi, 2021-12-05:
Vermutlich der nun behobene Fehler

N3m3515, 2021-12-05:
hatte es noch mit version 0.7.8 getestet.  seit 0.8 bin ich bei der selbst compilten version

closed 2021-12-06 by N3m3515

horchi, 2021-12-07:
kannst du mal auf mail kommen (box AT jwendel.de) dann können wir mal schauen ob wir das mit dem docker hingekommen.
Wenn es fertig ist posten wir es hier