horchi/linux-p4d #26: Problem beim Compilen mit libwebsocket

N3m3515, 2021-12-06:
Hallo,

ich versuche das Projekt gerade in ein Docker Image zu packen um das dann zur einfacheren Installation bereitstellen zu können.
Leider laufe ich in einen Fehler den ich dachte bereits schonmal gelöst zu haben:
`/usr/bin/ld: /usr/local/lib/libwebsockets.a(unix-caps.c.o): in function `_lws_plat_apply_caps':
unix-caps.c:(.text+0x24): undefined reference to `cap_get_proc'
/usr/bin/ld: unix-caps.c:(.text+0x44): undefined reference to `cap_set_flag'
/usr/bin/ld: unix-caps.c:(.text+0x4c): undefined reference to `cap_set_proc'
/usr/bin/ld: unix-caps.c:(.text+0x70): undefined reference to `cap_free'
collect2: error: ld returned 1 exit status
make: *** [Makefile:53: p4d] Error 1
`
Ich erstelle zuerst ein Base Image mit der folgenden Dockerfile:

`FROM balenalib/rpi-raspbian:buster

RUN apt-get update && apt-get install -y git build-essential pkg-config libssl-dev libjansson-dev libxml2-dev libcurl4-openssl-dev libssl-dev libmariadbclient-dev libmariadb-dev-compat uuid-dev cmake libcap-dev ssmtp mailutils haveged python3-dev wiringpi

RUN cd /usr/src/

RUN git clone https://libwebsockets.org/repo/libwebsockets libwebsockets

RUN cd libwebsockets && mkdir build && cd build && cmake .. && make -s && make -s install
`
Der Output beim compilen von libwebsocket sieht gut aus:

[https://pastebin.com/yJ1Kk5RH](https://pastebin.com/yJ1Kk5RH)

Danach nehme ich das Baseimage und gehe folgende Schritte in der Dockerfile durch:

`FROM linux_p4d-base

RUN cd /usr/src/

RUN git clone https://github.com/horchi/linux-p4d/

RUN cd linux-p4d/ && make clean all && make install

EXPOSE 1111
`

Da kann man sicher noch einiges optimieren aber ich lerne gerade noch :-)

Ich frage mich nur wieso er diesen Fehler beim compilen hat.

horchi, 2021-12-06:
Ich würde es nicht im docker bauen und ins docker Image auch keine Quellen und dev Pakete packen.
Baue das doch und installiere die fertigen binaries in das image?

N3m3515, 2021-12-06:
Ja das würde Sinn machen. also make clean all ausserhalb vom image und dann den ordner linux-p4d ind docker image und dann make install?

Welche abhängigkeiten brauche ich dann noch im image?

horchi, 2021-12-06:
Make ausserhalb.
Dann male install mit prefix Ordner in einen Pfad dessen wurzel du zum Image bauen nimmst.
Die Abhängigkeiten kannst du dir aus dem debian control file kopieren

N3m3515, 2021-12-06:
danach dann das postinstall script aufrufen?

horchi, 2021-12-06:
Welches postinstall?

N3m3515, 2021-12-06:
Ich habe jetzt mit make install DESTDIR=/home/pi/docker-linux-p4d/linux-p4d/binaries die binaries gebaut und kopiere die dann im makefile in mein baseimage:

COPY linux-p4d/binaries/ /

in dem binaries ordner sind aber folgende dateien:
[image]

hier zb den inhalt von postinstall:

https://pastebin.com/VCJZyV9e

muss ich ja sicher noch ausführen damit er den dienst registriert usw:?

horchi, 2021-12-06:
Das unter DEBIAN benötigst du nicht das ist nur zum bauen eines deb. Mein Hinweis darauf war lediglich um die Abhängigkeiten abzuschauen.
Ich schaue morgen wegen dem make install Aufruf da klemmt es.

N3m3515, 2021-12-06:
Ich habe die compileten binaries jetzt in nem image. aber ich hänge jetzt noch am entrypoint des baseimages. aber für heute ist mal schluss. habe gerade genug zeit darauf verschwendet die binaries zu starten ohne, dass der container einfach stoppt.

horchi, 2021-12-07:
kannst du mal auf mail kommen (box AT jwendel.de) dann können wir mal schauen ob wir das mit dem docker hingekommen.
Wenn es fertig ist posten wir es hier

N3m3515, 2021-12-08:
> ... ich hatte damit damals keinen Erfolg: https://github.com/ranseyer/p4d-docker/blob/main/Dockerfile (Ich brauche ein Image welches auch X86 kann!)
> 
> Falls es dich Interessiert könnte ich mal nachschauen was damals genau das Problem war. Meinen Ansatz solltest du nicht verfolgen. Ich habe auch beim Imagebau kompiliert weil es kein fertiges Paket für X86 gibt. (Und vom Canonical Launchpad habe ich auch nicht mehr wirklich viel Ahnung, sonst könnte man dort Mal sauber die Pakete für div. Plattformen bauen...)
> 
> PS: Ich denke dir fehlte die libcap-dev

Ich habe dich zu meinem Git eingeladen. Habe da jetzt ein funktionierendes Image für arm/raspi

closed 2021-12-08 by N3m3515