horchi/linux-p4d #29: Debian package

faxm0dem, 2022-02-03:
Could you please share your debian package scripts ?
The reason I'm asking is that I'd like to build my own debian package, without the `exim` dependency, as I'm using `postfix`.

Thank you

horchi, 2022-02-03:
Generally this software is provided in the form of source code.

The package is intended as a convinient installation option under Raspberry Pi OS Bullseye so that after the installation everything is available including the mail dispatch. 

I'll see if I can move the installation msmtp-mta from the package dependencies to the install script. If it will be compatible to different Debian versions I can't say and I can't test it. The focus of the package is "Raspberry Pi OS Bullseye".

horchi, 2022-02-03:
> Could you please share your debian package scripts ?
the build setup for the package is in the git: https://github.com/horchi/linux-p4d/tree/master/contrib/DEBIAN
or which script do you mean?

faxm0dem, 2022-02-03:
ah yes, you're absolutely right my apologies !

closed 2022-02-03 by faxm0dem