horchi/linux-p4d #32: How to disable MQTT ?

faxm0dem, 2022-02-04:
I tried to remove the line containing the mqtt address, but I still get the messages in syslog:

```
Feb 04 11:49:59 raspberrypi p4d[11557]: Error: MQTT connection broken, trying reconnect
Feb 04 11:49:59 raspberrypi p4d[11557]: Error: Failed to open socket 'Too many open files'
Feb 04 11:49:59 raspberrypi p4d[11557]: Error: MQTT: Connecting publisher to 'tcp://localhost:1883' failed
```

horchi, 2022-02-04:
can you please check what you get with 
`select * from config where name = 'mqttUrl';`

faxm0dem, 2022-02-04:
```
MariaDB [p4]> select * from config where name = 'mqttUrl';
+-------+---------+------------+------------+-------+
| owner | name    | inssp      | updsp      | value |
+-------+---------+------------+------------+-------+
| p4d   | mqttUrl | 1643875973 | 1643971734 |       |
+-------+---------+------------+------------+-------+
1 row in set (0.002 sec)
```

faxm0dem, 2022-02-04:
I think I may have forgotten to restart `p4d` : now it seems it is not complaining anymore about mqtt connection.

horchi, 2022-02-04:
okay fine.
I will check why the config not applied without a restart

faxm0dem, 2022-02-04:
By the way, the connection with mosquitto doesn't work out of the box on bullseye : apparently the anonymous user is disabled and also the server doesn't listen on localhost anymore.

I tried to change it by adding the following config to mosquitto:

```
listener 1883 127.0.0.1
allow_anonymous true
```

But it didn't help:

```
Feb  3 11:20:27 raspberrypi p4d[11091]: Error: MQTT connection broken, trying reconnect
Feb  3 11:20:27 raspberrypi p4d[11091]: Error: mqtt_sync for connection '' failed, result was -2147483631 'MQTT_ERROR_SOCKET_ERROR'
Feb  3 11:20:27 raspberrypi p4d[11091]: Error: Can't subscribe, not connected or topic 'p4d2mqtt/command/#' missing
```

That's why I wanted to disable mqtt for now.

horchi, 2022-02-05:
I cant reproduce, after a fresh bullseye installation I installed mosquitto.

Console 1:
`mosquitto_sub -v -t test`

Console 2:
`mosquitto_pub -t test -m foo`

then on console 1 appears:  
`test foo`

horchi, 2022-02-06:
is there a special conf on your system after installing buster and mosqiutoo from scratch:
`ls -l   /etc/mosquitto/conf.d/`
or something special here:
```
root@rpi4:~# cat   /etc/mosquitto/mosquitto.conf 
# Place your local configuration in /etc/mosquitto/conf.d/
# /usr/share/doc/mosquitto/examples/mosquitto.conf.example
pid_file /run/mosquitto/mosquitto.pid
persistence true
persistence_location /var/lib/mosquitto/
log_dest file /var/log/mosquitto/mosquitto.log
include_dir /etc/mosquitto/conf.d
```
I'am on this system:
```
root@rpi4:~# lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```

horchi, 2022-02-06:
I also double checked the reload/apply of the MQTT configuration. If I clear the config line `MQTT Broker Url` and click i get this in the log file (if the MQTT log option is activated):
`Feb  6 10:01:57 gate p4d: Disconnected from MQTT`
an than no more connection trials.

In the next version I will add this log message additional:
`Feb  6 10:01:57 gate p4d: Config of MQTT url changed from 'tcp://192.168.200.101:1883' to '', disconnecting`
only to get some more informations

faxm0dem, 2022-02-08:
I didn't change the mosquitto config, but I'm new to mosquitto so I'm probably doing something wrong.
Thanks so much for looking into this, and for the additional log entry.

I tried your pubsub example and it works.
How do I configure topics on the p4d side ?

faxm0dem, 2022-02-08:
Some additional info:

```
# netstat -tapn|grep 1883
tcp        0      0 127.0.0.1:1883          0.0.0.0:*               LISTEN      7940/mosquitto
tcp        0      0 127.0.0.1:34634         127.0.0.1:1883          ESTABLISHED 7993/p4d
tcp        0      0 127.0.0.1:34628         127.0.0.1:1883          TIME_WAIT   -
tcp        0      0 127.0.0.1:34636         127.0.0.1:1883          ESTABLISHED 7993/p4d
tcp        0      0 127.0.0.1:34626         127.0.0.1:1883          TIME_WAIT   -
tcp        0      0 127.0.0.1:1883          127.0.0.1:34634         ESTABLISHED 7940/mosquitto
tcp        0      0 127.0.0.1:1883          127.0.0.1:34636         ESTABLISHED 7940/mosquitto
```

horchi, 2022-02-08:
looks strange, it seems connected yet error messages. The configuration looks good, even without topics this should not happen.
Does the mosquitto process run permanently or does it always crash and get restarted by syslogd.

faxm0dem, 2022-02-08:
I tested your commands : they work perfectly.
The `mosquitto` daemon doesn't crash, it's also pretty silent (syslog-wise)

horchi, 2022-02-08:
is there something more around this message: `Feb 08 09:50:48 raspberrypi p4d[7993]: Error: MQTT connection broken, trying reconnect` ?
Did you install the p4d from package or by building from source? Which version you have?

faxm0dem, 2022-02-08:
I built from source no other syslig msgs

horchi, 2022-02-09:
I can't explain the effect and unfortunately I can't reproduce it. 
Wernli sounds like Switzerland, do you speak German?
I could offer you to look at it in a TeamViewer session, maybe I can see something there.

faxm0dem, 2022-02-09:
Yes I speak German, it's very kind of you to try to help.
I will continue to try to find out what the issue is, and if I can't figure it out, we'll continue on the phone :-)

horchi, 2022-02-17:
please update to 0.9.31, I found the a possible reason which should be shown in the log if you activate the MQTT log option:
`No update from MQTT since '%s', disconnect from MQTT to force recover`
appear if the w1mqtt process is not started or even not working properly.
In p4d version 0.9.31 this should be fixed and even without the w1mqtt no unnecessary cyclic reconnect.

faxm0dem, 2022-02-18:
Thanks I will test this ASAP

horchi, 2022-02-18:
and please with the 0.9.32 ;)

horchi, 2022-05-03:
seems to work, I'll close the issue once

closed 2022-05-03 by horchi