horchi/linux-p4d #35: Installation of 0.9.40 not working -  failing in websock.c

enr0c, 2022-10-23:
`websock.c: In member function ‘int cWebSock::init(int, int, const char*, bool)’:
websock.c:70:17: error: ‘struct lws_protocols’ has no member named ‘tx_packet_size’; did you mean ‘rx_buffer_size’?
    protocols[1].tx_packet_size = 0;
                 ^~~~~~~~~~~~~~
                 rx_buffer_size
websock.c:95:10: error: ‘struct lws_http_mount’ has no member named ‘basic_auth_login_file’
    mount.basic_auth_login_file = nullptr;
          ^~~~~~~~~~~~~~~~~~~~~
websock.c:131:9: error: ‘struct lws_context_creation_info’ has no member named ‘ws_ping_pong_interva ’
    info.ws_ping_pong_interval = timeout;
         ^~~~~~~~~~~~~~~~~~~~~
websock.c: In function ‘void getClientInfo(lws*, std::__cxx11::string*)’:
websock.c:279:7: error: ‘lws_get_peer_simple’ was not declared in this scope
       lws_get_peer_simple(wsi, clientName, sizeof(clientName));
       ^~~~~~~~~~~~~~~~~~~
websock.c:279:7: note: suggested alternative: ‘BIO_get_peer_name’
       lws_get_peer_simple(wsi, clientName, sizeof(clientName));
       ^~~~~~~~~~~~~~~~~~~
       BIO_get_peer_name
websock.c: In static member function ‘static int cWebSock::callbackHttp(lws*, lws_callback_reasons, void*, void*, size_t)’:
websock.c:430:12: error: ‘LWS_CALLBACK_HTTP_BIND_PROTOCOL’ was not declared in this scope
       case LWS_CALLBACK_HTTP_BIND_PROTOCOL:       // 49
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
websock.c:430:12: note: suggested alternative: ‘LWS_CALLBACK_HTTP_WRITEABLE’
       case LWS_CALLBACK_HTTP_BIND_PROTOCOL:       // 49
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            LWS_CALLBACK_HTTP_WRITEABLE
websock.c:431:12: error: ‘LWS_CALLBACK_HTTP_DROP_PROTOCOL’ was not declared in this scope
       case LWS_CALLBACK_HTTP_DROP_PROTOCOL:       // 50
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
websock.c:431:12: note: suggested alternative: ‘LWS_CALLBACK_HTTP_WRITEABLE’
       case LWS_CALLBACK_HTTP_DROP_PROTOCOL:       // 50
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            LWS_CALLBACK_HTTP_WRITEABLE
websock.c: In static member function ‘static int cWebSock::callbackWs(lws*, lws_callback_reasons, void*, void*, size_t)’:
websock.c:564:28: error: ‘lws_write_ws_flags’ was not declared in this scope
                int flags = lws_write_ws_flags(LWS_WRITE_TEXT, !clients[wsi].msgBufferSendOffset, pending <= maxChunk);
                            ^~~~~~~~~~~~~~~~~~
websock.c:564:28: note: suggested alternative: ‘lws_write_protocol’
                int flags = lws_write_ws_flags(LWS_WRITE_TEXT, !clients[wsi].msgBufferSendOffset, pending <= maxChunk);
                            ^~~~~~~~~~~~~~~~~~
                            lws_write_protocol
websock.c:599:17: error: ‘lws_is_first_fragment’ was not declared in this scope
             if (lws_is_first_fragment(wsi))
                 ^~~~~~~~~~~~~~~~~~~~~
websock.c:599:17: note: suggested alternative: ‘lws_is_final_fragment’
             if (lws_is_first_fragment(wsi))
                 ^~~~~~~~~~~~~~~~~~~~~
                 lws_is_final_fragment
websock.c:695:12: error: ‘LWS_CALLBACK_ADD_HEADERS’ was not declared in this scope
       case LWS_CALLBACK_ADD_HEADERS:
            ^~~~~~~~~~~~~~~~~~~~~~~~
websock.c:695:12: note: suggested alternative: ‘LWS_CALLBACK_ADD_POLL_FD’
       case LWS_CALLBACK_ADD_HEADERS:
            ^~~~~~~~~~~~~~~~~~~~~~~~
            LWS_CALLBACK_ADD_POLL_FD
websock.c: In static member function ‘static int cWebSock::getIntParameter(lws*, const char*, int)’:
websock.c:818:22: error: ‘lws_get_urlarg_by_name’ was not declared in this scope
    const char* arg = lws_get_urlarg_by_name(wsi, name, buf, 100);
                      ^~~~~~~~~~~~~~~~~~~~~~
websock.c:818:22: note: suggested alternative: ‘lws_get_parent’
    const char* arg = lws_get_urlarg_by_name(wsi, name, buf, 100);
                      ^~~~~~~~~~~~~~~~~~~~~~
                      lws_get_parent
websock.c: In static member function ‘static const char* cWebSock::getStrParameter(lws*, const char*, const char*)’:
websock.c:830:22: error: ‘lws_get_urlarg_by_name’ was not declared in this scope
    const char* arg = lws_get_urlarg_by_name(wsi, name, buf, 512);
                      ^~~~~~~~~~~~~~~~~~~~~~
websock.c:830:22: note: suggested alternative: ‘lws_get_parent’
    const char* arg = lws_get_urlarg_by_name(wsi, name, buf, 512);
                      ^~~~~~~~~~~~~~~~~~~~~~
                      lws_get_parent`

Can you please help ?

horchi, 2022-10-23:
what version of libwebsockets are you using?

enr0c, 2022-10-23:
Thanks for the prompt reaction. The following version is installed:

Package: libwebsockets-dev
Version: 4.0.20-2
Priority: optional
Section: libdevel
Source: libwebsockets
Maintainer: Laszlo Boszormenyi (GCS) <gcs@debian.org>
Installed-Size: 1.356 kB
Depends: libwebsockets16 (= 4.0.20-2), libev-dev, libssl-dev, libuv1-dev, zlib1g-dev, libcap-dev
Homepage: https://libwebsockets.org/
Download-Size: 399 kB
APT-Sources: http://raspbian.raspberrypi.org/raspbian bullseye/main armhf Packages
Description: lightweight C websockets library - development files
 Libwebsockets is a lightweight pure C library for both websockets
 clients and servers built to use minimal CPU and memory resources
 and provide fast throughput in both directions.
 .
 This package contains the header files needed for developing programs
 using libwebsockets and a static library.

Package: libwebsockets-dev
Version: 2.0.3-3
Priority: optional
Section: libdevel
Source: libwebsockets
Maintainer: Laszlo Boszormenyi (GCS) <gcs@debian.org>
Installed-Size: 417 kB
Depends: libwebsockets8 (= 2.0.3-3), libev-dev, libssl-dev, libuv1-dev, zlib1g-dev
Homepage: https://libwebsockets.org/
Download-Size: 144 kB
APT-Manual-Installed: yes
APT-Sources: http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
Description: lightweight C websockets library - development files
 Libwebsockets is a lightweight pure C library for both websockets
 clients and servers built to use minimal CPU and memory resources
 and provide fast throughput in both directions.
 .
 This package contains the header files needed for developing programs
 using libwebsockets and a static library.

horchi, 2022-10-23:
try the minimal version of the library mentioned in README

enr0c, 2022-10-23:
That solved the issue. Many thanks!
`cd usr/src
git clone https://libwebsockets.org/repo/libwebsockets
cd libwebsockets
mkdir build
cd build
cmake ..
make
sudo make install`

closed 2022-10-23 by enr0c