horchi/linux-p4d #41: MQTT Error with HomeAssistant

Tropaion, 2023-08-04:
Hello @horchi,
I love your project und use it since it started, but ever since I switched from FHEM to HomeAssitant, I have many problems and I hope you can help me.
First, string values do not work in HA, I get following error:
```
2023-08-04 18:41:08.158 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.heizung_zustand, topic: 'p4d2mqtt/Heizung/Status/state' with payload: b'{"value": "Brenner Aus"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 593, in state
    numerical_value = float(value)  # type:ignore[arg-type]
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: 'Brenner Aus'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 746, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 808, in _async_write_ha_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 752, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 595, in state
    raise ValueError(
ValueError: Sensor sensor.heizung_zustand has device class 'None', state class 'None' unit '' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'Brenner Aus' ()
```
* I get topics/sensor I never configured:
[image]
* Is there a way to configure it, so that HA MQTT Discovery groups the entities into one single device? Currently when discovered, the entities are not grouped together.
* My last problem, is there a way to get the error messages sent via MQTT? A few years ago when I setup FHEM, I got them somehow but can't remember how I did it.

hoppel118, 2023-09-01:
Hi,

same story here. Moved from FHEM to Home Assistant at the beginning of the year. It worked as expected, but now I also the message:

```
Log Details (ERROR)

Logger: homeassistant.components.mqtt.models
Source: components/sensor/__init__.py:595 
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22)) 
First occurred: August 24, 2023 at 11:14:59 PM (80104 occurrences) 
Last logged: 10:03:06 PM

Exception raised when updating state of sensor.heizungsstatus_heizung, topic: 'p4d2mqtt/sensor/Status/state' with payload: b'{"value": "Feuer Aus"}'
Exception raised when updating state of sensor.betriebsart_kessel_heizung, topic: 'p4d2mqtt/sensor/Betriebsart_Kessel/state' with payload: b'{"value": "Dauerlast"}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 593, in state
    numerical_value = float(value)  # type:ignore[arg-type]
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: 'Feuer Aus'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 746, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 808, in _async_write_ha_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 752, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 595, in state
    raise ValueError(
ValueError: Sensor sensor.heizungsstatus_heizung has device class 'None', state class 'None' unit '' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'Feuer Aus' (<class 'str'>)
```

I am updating my p4d running on an rpi via the horchi‘s script.

@horchi Can you please integrate the code change from @helmut8960 ?

Thank you so much!

Hoppel

Tropaion, 2023-09-01:
Hi @hoppel118,
I looks like @horchi is since a while, inactive.
As temporary solution you can install from my fork.

hoppel118, 2023-09-02:
Hi @Tropaion,

can you describe the installation procedure, please?

Thanks Hoppel

Tropaion, 2023-09-02:
Something like this: https://wiki.flowerhouse.at/index.php/Heater
The relevant parts are:
* 1.1	libwebsockets
* 1.2	p4d daemon
If you install on raspberry, you have to remove `NO_RASPBERRY_PI=1`. 
And you have to download from my fork.

horchi, 2023-12-24:
In the medium term (as soon as I find the time) I plan to somehow outsource the support for various automations from the project. I don't know exactly what that will look like at the moment. The goal is to no longer have to support x different automations, especially not the HomeAssistant, which seems to be the least flexible of all the ones I deal with. Another reason is that I no longer use any of these automations myself and therefore have no way of testing the interface. I have been using https://github.com/horchi/linux-homectld for years now, which was developed on the basis of p4d.

closed 2023-12-29 by horchi