horchi/linux-p4d #44: MQTT Werte mit Value davor

raue80, 2023-10-27:
Hallo zusammen, 
das Projekt ist echt super und läuft bei mir schon seit 2016. Nun habe ich den IObroker mit MQTT angebunden. 
Leider stehen vor den Werten immer value. Wie bekomme ich das weg?
Danke tolle Arbeit. 
[image]

horchi, 2023-10-27:
Hi,  

aus reiner Neugier, woran klag es das du gar keine Daten bekommen hats, habe gerade den geschlossenen Issue entdeckt.

Zu dem Fall hier, ich sehe kein value vor den Wert, kannst du mir mal die Plain JSON Ausgabe posten? Die bekommst du zum Beispiel von `mosquitto_sub -v -t <topic>`

raue80, 2023-10-27:
Hi, danke für die schnelle Rückmeldung.
Ich kann es nicht einmal richtig nachvollziehen. Mehrmaliges Installieren des MQTT Adapter auf dem iobroker und gefühlt 100 Neustarts des p4d Services auf dem raspi haben zum Erfolg geführt.
Ich hatte immer die Meldung, dass der p4d2mqtt/ping erfolgreich ist, aber bei p4d2mqtt/command/# kam ein Fehler ??
Die json poste ich später, wenn ich Feierabend habe

horchi, 2023-10-27:
> Ich hatte immer die Meldung, dass der p4d2mqtt/ping erfolgreich ist, aber bei p4d2mqtt/command/# kam ein Fehler ??
merkwürdig

raue80, 2023-10-27:
Ja schon.
Die Werte kommen nun rüber, nur das vorangestellte "value": vor dem Wert ist komisch
Wie gesagt ich poste später das json
Danke schon einmal für die Hilfe

raue80, 2023-10-27:
Hi, so das ist nun, was rauskommt:
[image]

horchi, 2023-10-29:
das schaut doch gut aus, es ist ein JSON Objekt mit genau einem Element, dem Wert und laut json_pp auch kein Fehler in in Formatierung/Syntax:
```
#>echo '{ "value": 55.3}' | json_pp
{
   "value" : 55.3
}

```
Du hats beim p4d demnach den <NAME> im Topic Namen, mimst du <GROUP>oder gar kein template in den Namen bekommst du ale Sensoren mit viel mehr Details für die Weiterverarbeitung auf ein Topic, das schaut dann so aus:

```
p4d2mqtt/p4/Heizung {"type": "P4UD", "address": 1, "unit": "zst", "title": "Status", "kind": "value", "text": "Kessel  Aus ", "image": "img/type/heating-P4.png"}
p4d2mqtt/p4/Heizung {"type": "P4UD", "address": 1, "unit": "zst", "title": "Status", "kind": "value", "text": "Kessel  Aus ", "image": "img/type/heating-P4.png"}
p4d2mqtt/p4/Heizung {"type": "P4UD", "address": 1, "unit": "zst", "title": "Status", "kind": "value", "text": "Kessel  Aus ", "image": "img/type/heating-P4.png"}
p4d2mqtt/p4/Heizung {"type": "P4W1", "address": 90673985, "unit": "°C", "title": "Puffer Mitte", "kind": "value", "value": 50.25}
```

bzw. Schön formatiert:
```
{
   "address" : 1,
   "image" : "img/type/heating-P4.png",
   "kind" : "value",
   "text" : "Kessel  Aus ",
   "title" : "Status",
   "type" : "P4UD",
   "unit" : "zst"
}
```
oder 
```
{
   "address" : 90673985,
   "kind" : "value",
   "title" : "Puffer Mitte",
   "type" : "P4W1",
   "unit" : "Â°C",
   "value" : 50.25
}

```
Die Option für ein Topic je Sensor ist für Systeme wie den Homeassistant mit welche man ordentliches JSON nicht oder nur schwer auswerten kann. Das würde ich am liebsten ganz ausbauen. Es ist nicht mein Ziel alle Sonderlocken für Potentiell angeschossene Systeme zu unterstützen. Dazu kann man mit ein nachgeschaltetes Script oder aber auch Node-Red und vieles mehr verwenden.

Ich persönlich bin nachdem ich eine Zeitlang den Homeassistant verwendet habe und mir vieles andere angesehen habe für die Haus-, Pool- und Wohnmobil Automatisierung ganz auf meine C++ Implementierung umgestiegen. Dazu hab ich in Prinzip  einen Clone vom p4d nur ohne den Programmteil zur Anbindung der Heizung

raue80, 2023-10-30:
Hallo,
danke für die schnelle Rückmeldung 
Ich habe es hierdurch lösen können

closed 2023-10-30 by raue80