horchi/linux-p4d #45: ‘time’ was not declared in this scope

Fabi00815, 2023-11-24:
Tryed no install it on a new Raspberry Pi 5 8GB with newest OS Bookworm:

g++ -c -ggdb -O0 -pthread -fPIC -std=c++17 -Wall -Wreturn-type -Wformat -pedantic -Wunused-variable -Wunused-label -Wunused-value -Wunused-function -Wno-long-long -Wno-unused-parameter -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE64_SOURCE -Wno-psabi -I/usr/include/mariadb -I/usr/include/mariadb/mysql -D_GNU_SOURCE -DTARGET='"p4d"' -DCLASS='P4d' -DNAME='"p4"' -DLOG_PREFIX='""' -DUSEUUID -DUSEMD5 -DUSELIBARCHIVE -DUSEJSON -DUSEGUNZIP -DUSEDB -DBIN_PATH='"/usr/local/bin"' -D_NO_RASPBERRY_PI_ -DGIT_REV='"eacaf7c"' -o lib/db.o lib/db.c
In file included from lib/db.h:23,
                 from lib/db.c:13:
lib/common.h: In member function ‘void MemoryStruct::clear()’:
lib/common.h:264:20: error: ‘time’ was not declared in this scope
  264 |          modTime = time(0);
      |                    ^~~~
lib/common.h:23:1: note: ‘time’ is defined in header ‘<ctime>’; did you forget to ‘#include <ctime>’?
   22 | #include <vector>
  +++ |+#include <ctime>
   23 | #include <list>
lib/db.h: In member function ‘void cDbValue::setValue(long int)’:
lib/db.h:222:23: error: aggregate ‘tm tm’ has incomplete type and cannot be defined
  222 |             struct tm tm;
      |                       ^~
lib/db.h:227:13: error: ‘localtime_r’ was not declared in this scope; did you mean ‘locale_t’?
  227 |             localtime_r(&v, &tm);
      |             ^~~~~~~~~~~
      |             locale_t
lib/db.h: In member function ‘time_t cDbValue::getTimeValue()’:
lib/db.h:360:20: error: aggregate ‘tm tm’ has incomplete type and cannot be defined
  360 |          struct tm tm;
      |                    ^~
lib/db.h:372:17: error: ‘mktime’ was not declared in this scope
  372 |          return mktime(&tm);
      |                 ^~~~~~
lib/db.h:25:1: note: ‘mktime’ is defined in header ‘<ctime>’; did you forget to ‘#include <ctime>’?
   24 | #include "dbdict.h"
  +++ |+#include <ctime>
   25 | 
lib/db.h: In constructor ‘cDbStatements::cDbStatements()’:
lib/db.h:519:44: error: ‘time’ was not declared in this scope
  519 |       cDbStatements()  { statisticPeriod = time(0); }
      |                                            ^~~~
lib/db.h:519:44: note: ‘time’ is defined in header ‘<ctime>’; did you forget to ‘#include <ctime>’?
lib/db.h: In member function ‘void cDbStatements::showStat(const char*)’:
lib/db.h:527:80: error: ‘time’ was not declared in this scope
  527 |          tell(eloAlways, "Statement statistic of last %ld seconds from '%s':", time(0) - statisticPeriod, name);
      |                                                                                ^~~~
lib/db.h:527:80: note: ‘time’ is defined in header ‘<ctime>’; did you forget to ‘#include <ctime>’?
lib/db.c: In member function ‘virtual int cDbTable::insert(time_t)’:
lib/db.c:1583:40: error: ‘time’ was not declared in this scope
 1583 |          setValue(fld, inssp ? inssp : time(0));
      |                                        ^~~~
lib/db.c:14:1: note: ‘time’ is defined in header ‘<ctime>’; did you forget to ‘#include <ctime>’?
   13 | #include "db.h"
  +++ |+#include <ctime>
   14 | 
lib/db.c: In member function ‘virtual int cDbTable::update(time_t)’:
lib/db.c:1616:40: error: ‘time’ was not declared in this scope
 1616 |          setValue(fld, updsp ? updsp : time(0));
      |                                        ^~~~
lib/db.c:1616:40: note: ‘time’ is defined in header ‘<ctime>’; did you forget to ‘#include <ctime>’?
make: *** [Make.config:110: lib/db.o] Fehler 1

enr0c, 2023-12-23:
I confirm this issue for the Bookworm OS

enr0c, 2023-12-23:
After adding

#include <ctime> in script lib/common.h

the installer fails here:

`
lib/thread.c: In constructor ‘cThread::cThread(const char*, bool)’:
lib/thread.c:66:3: error: ‘pthread_attr_init’ was not declared in this scope; did you mean ‘pthread_attr_t’?
   66 |   pthread_attr_init(&attr);
      |   ^~~~~~~~~~~~~~~~~
      |   pthread_attr_t
lib/thread.c: In destructor ‘virtual cThread::~cThread()’:
lib/thread.c:73:3: error: ‘pthread_attr_destroy’ was not declared in this scope; did you mean ‘pthread_attr_t’?
   73 |   pthread_attr_destroy(&attr);
      |   ^~~~~~~~~~~~~~~~~~~~
      |   pthread_attr_t
lib/thread.c: In member function ‘bool cThread::Start(int, int)’:
lib/thread.c:166:18: error: ‘pthread_create’ was not declared in this scope; did you mean ‘timer_create’?
  166 |            res = pthread_create(&childTid, 0, (void*(*)(void*))&StartThread, (void*)this);
      |                  ^~~~~~~~~~~~~~
      |                  timer_create
lib/thread.c:170:12: error: ‘pthread_attr_setstacksize’ was not declared in this scope
  170 |            pthread_attr_setstacksize(&attr, stackSize);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~
lib/thread.c:171:18: error: ‘pthread_create’ was not declared in this scope; did you mean ‘timer_create’?
  171 |            res = pthread_create(&childTid, &attr, (void*(*)(void*))&StartThread, (void*)this);
      |                  ^~~~~~~~~~~~~~
      |                  timer_create
lib/thread.c:176:12: error: ‘pthread_detach’ was not declared in this scope; did you mean ‘pthread_t’?
  176 |            pthread_detach(childTid); // auto-reap
      |            ^~~~~~~~~~~~~~
      |            pthread_t
lib/thread.c: In member function ‘void cThread::Cancel(int)’:
lib/thread.c:230:6: error: ‘pthread_cancel’ was not declared in this scope; did you mean ‘pthread_once_t’?
  230 |      pthread_cancel(childTid);
      |      ^~~~~~~~~~~~~~
      |      pthread_once_t
lib/thread.c: In constructor ‘cCondWait::cCondWait()’:
lib/thread.c:248:4: error: ‘pthread_mutex_init’ was not declared in this scope; did you mean ‘pthread_mutex_t’?
  248 |    pthread_mutex_init(&mutex, NULL);
      |    ^~~~~~~~~~~~~~~~~~
      |    pthread_mutex_t
lib/thread.c:249:4: error: ‘pthread_cond_init’ was not declared in this scope; did you mean ‘pthread_cond_t’?
  249 |    pthread_cond_init(&cond, NULL);
      |    ^~~~~~~~~~~~~~~~~
      |    pthread_cond_t
lib/thread.c: In destructor ‘cCondWait::~cCondWait()’:
lib/thread.c:254:4: error: ‘pthread_cond_broadcast’ was not declared in this scope; did you mean ‘pthread_condattr_t’?
  254 |    pthread_cond_broadcast(&cond); // wake up any sleepers
      |    ^~~~~~~~~~~~~~~~~~~~~~
      |    pthread_condattr_t
lib/thread.c:255:4: error: ‘pthread_cond_destroy’ was not declared in this scope; did you mean ‘pthread_condattr_t’?
  255 |    pthread_cond_destroy(&cond);
      |    ^~~~~~~~~~~~~~~~~~~~
      |    pthread_condattr_t
lib/thread.c:256:4: error: ‘pthread_mutex_destroy’ was not declared in this scope; did you mean ‘pthread_mutexattr_t’?
  256 |    pthread_mutex_destroy(&mutex);
      |    ^~~~~~~~~~~~~~~~~~~~~
      |    pthread_mutexattr_t
lib/thread.c: In member function ‘bool cCondWait::Wait(int)’:
lib/thread.c:267:4: error: ‘pthread_mutex_lock’ was not declared in this scope; did you mean ‘pthread_mutex_t’?
  267 |    pthread_mutex_lock(&mutex);
      |    ^~~~~~~~~~~~~~~~~~
      |    pthread_mutex_t
lib/thread.c:279:20: error: ‘pthread_cond_timedwait’ was not declared in this scope; did you mean ‘pthread_condattr_t’?
  279 |                if (pthread_cond_timedwait(&cond, &mutex, &abstime) == ETIMEDOUT)
      |                    ^~~~~~~~~~~~~~~~~~~~~~
      |                    pthread_condattr_t
lib/thread.c:285:10: error: ‘pthread_cond_wait’ was not declared in this scope; did you mean ‘pthread_cond_t’?
[... 26 lines cut ...]
lib/thread.c:314:4: error: ‘pthread_cond_broadcast’ was not declared in this scope; did you mean ‘pthread_condattr_t’?
  314 |    pthread_cond_broadcast(&cond); // wake up any sleepers
      |    ^~~~~~~~~~~~~~~~~~~~~~
      |    pthread_condattr_t
lib/thread.c:315:4: error: ‘pthread_cond_destroy’ was not declared in this scope; did you mean ‘pthread_condattr_t’?
  315 |    pthread_cond_destroy(&cond);
      |    ^~~~~~~~~~~~~~~~~~~~
      |    pthread_condattr_t
lib/thread.c: In member function ‘void cCondVar::Wait(cMyMutex&)’:
lib/thread.c:325:7: error: ‘pthread_cond_wait’ was not declared in this scope; did you mean ‘pthread_cond_t’?
  325 |       pthread_cond_wait(&cond, &Mutex.mutex);
      |       ^~~~~~~~~~~~~~~~~
      |       pthread_cond_t
lib/thread.c: In member function ‘bool cCondVar::TimedWait(cMyMutex&, int)’:
lib/thread.c:347:14: error: ‘pthread_cond_timedwait’ was not declared in this scope; did you mean ‘pthread_condattr_t’?
  347 |          if (pthread_cond_timedwait(&cond, &Mutex.mutex, &abstime) == ETIMEDOUT)
      |              ^~~~~~~~~~~~~~~~~~~~~~
      |              pthread_condattr_t
lib/thread.c: In member function ‘void cCondVar::Broadcast()’:
lib/thread.c:359:4: error: ‘pthread_cond_broadcast’ was not declared in this scope; did you mean ‘pthread_condattr_t’?
  359 |    pthread_cond_broadcast(&cond);
      |    ^~~~~~~~~~~~~~~~~~~~~~
      |    pthread_condattr_t
make: *** [Make.config:110: lib/thread.o] Error 1
`

horchi, 2023-12-24:
I now added include for the time and pthread header - version  0.9.49
Can you please try again (I don't have a bookworm installation yet)?

enr0c, 2024-01-02:
i confirm, installation works. issue fixed.
Thanks a lot!

horchi, 2024-01-02:
thx for testing

closed 2024-01-02 by horchi