horchi/linux-p4d #46: Operating modes and conditions are not human-readable

hoppel118, 2024-01-02:
P4D: 0.9.53-GITf7e96b1
HA Core: 2023.12.4
HA Supervisor: 2023.12.0
HA Operating System: 11.2
HA Frontend: 20231208.2

Hi, 

happy new year.

Some of the operating mode (Betriebsart) and condition (Zustand) entities are not resolved into human-readable words in p4d (and Home Assistant).

I only see, 1.0, 2.0 and so on:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

In Home Assistant it looks as follows for example:

[image]
[image]
[image]

@Tropaion Does this work with your fork or do you have an idea why it behaves like that? What is needed to analyse this?

Thanks and regards Hoppel

Tropaion, 2024-01-03:
I don't use "Betriebsart Kessel" so I can't say, but the entities I use don't really have problems:
[image]

horchi, 2024-01-11:
can you please describe the problem again using the display of the p4d WEB interface if it is still open.
Or has it been resolved in the meantime and I can close the issue?

hoppel118, 2024-01-11:
Hi @horchi, 

sorry, for the late response. I do not have much time at the moment. What I would like to see is, that the following values get translated from numerical values to human readable values:

[image]
[image]
[image]
[image]
[image]

like it is already the case for for the following value:

[image]

for example:
- the value of "Betriebsart Kessel" is human readable: Dauerlast
- the value of "Pufferladezustand" is human readable: 36 (%)
- all the other values only show 0.00, 1.00, 2.00, 3.00 and so on ...

These values already aren't human readable in your p4d web interface. So, forget all the Infos about home assistant above. I think, this has to be solved on your side, even though I am unsure, if it ever worked.

I hope you now understand, what I mean. :)

Regards Hoppel

horchi, 2024-01-11:
kannst du mir mal bitte die Adressen geben, ich finde die Werte bei mir nicht, ggf finde ich sie über die Adressen

hoppel118, 2024-01-11:
Ich hoffe, das sind die Adressen die du meinst:

Dies funktioniert jetzt schon korrekt:

- Betriebsart Kessel UD:0x02
- Pufferladezustand

Dies nicht:

- Betriebsart Heizkreis VA:0x17
- Betriebsart Heizkreis VA:0x1b
- Betriebsart Boiler VA:0x156
- Pufferladezustand Diskret VA:0x1c0
- Kesselzustand VA:0x1c9

horchi, 2024-01-11:
die beiden "Betriebsart Heizkreis" sind bei mir "Partyschalter", die anderen Adressen kennt meine S32ß00 gar nicht.
Kannst du mal das für die Parameter machen, mal sehen ob da ein Text geliefert wird:
`p4 getv -l 15 -a 0x17`

horchi, 2024-01-22:
@hoppel118 klappt das?

hoppel118, 2024-01-22:
Sorry @horchi, dass ich mich jetzt erst melde. Habe momentan wirklich viel zu tun. Danke für den Reminder. ;)

```
p4 getv -l 15 -a 0x17
13:37:21,882 Opening '/dev/ttyUSB0' succeeded!
13:37:21,986 -> 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:37:21,993 got 02
13:37:21,993 got FD
13:37:21,993 got 00
13:37:21,993 got 0B
13:37:21,993 got 22
13:37:21,993 got 54
13:37:21,993 got 65
13:37:21,993 got 73
13:37:21,993 got 63
13:37:21,993 got 68
13:37:21,994 got 74
13:37:21,994 got 20
13:37:21,994 got 3B
13:37:21,994 got 2D
13:37:21,994 got 29
13:37:21,994 got 1C
13:37:21,994 <- 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:37:22,005 -> 02 FD 00 03 30 00 17 6D    ....0..m
13:37:22,009 got 02
13:37:22,009 got FD
13:37:22,009 got 00
13:37:22,009 got 03
13:37:22,009 got 30
13:37:22,009 got 00
13:37:22,009 got 00
13:37:22,009 got 54
13:37:22,009 <- 02 FD 00 03 30 00 00 54    ....0..T
13:37:22,020 value 0x17 is 0
13:37:22,020 Closing io device
```

```
p4 getv -l 15 -a 0x1b
13:38:29,148 Opening '/dev/ttyUSB0' succeeded!
13:38:29,252 -> 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:38:29,270 got 02
13:38:29,270 got FD
13:38:29,270 got 00
13:38:29,270 got 0B
13:38:29,271 got 22
13:38:29,271 got 54
13:38:29,271 got 65
13:38:29,271 got 73
13:38:29,271 got 63
13:38:29,271 got 68
13:38:29,271 got 74
13:38:29,271 got 20
13:38:29,271 got 3B
13:38:29,271 got 2D
13:38:29,271 got 29
13:38:29,271 got 1C
13:38:29,271 <- 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:38:29,282 -> 02 FD 00 03 30 00 1B 79    ....0..y
13:38:29,286 got 02
13:38:29,286 got FD
13:38:29,286 got 00
13:38:29,286 got 03
13:38:29,286 got 30
13:38:29,287 got 00
13:38:29,287 got 00
13:38:29,287 got 54
13:38:29,287 <- 02 FD 00 03 30 00 00 54    ....0..T
13:38:29,298 value 0x1b is 0
13:38:29,298 Closing io device
```

```
p4 getv -l 15 -a 0x156
13:39:01,414 Opening '/dev/ttyUSB0' succeeded!
13:39:01,518 -> 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:39:01,536 got 02
13:39:01,536 got FD
13:39:01,537 got 00
13:39:01,537 got 0B
13:39:01,537 got 22
13:39:01,538 got 54
13:39:01,538 got 65
13:39:01,538 got 73
13:39:01,538 got 63
13:39:01,538 got 68
13:39:01,538 got 74
13:39:01,538 got 20
13:39:01,539 got 3B
13:39:01,539 got 2D
13:39:01,539 got 29
13:39:01,539 got 1C
13:39:01,540 <- 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:39:01,551 -> 02 FD 00 03 30 01 56 AD    ....0.V.
13:39:01,568 got 02
13:39:01,569 got FD
13:39:01,569 got 00
13:39:01,569 got 03
13:39:01,569 got 30
13:39:01,570 got 00
13:39:01,570 got 01
13:39:01,570 got 57
13:39:01,571 <- 02 FD 00 03 30 00 01 57    ....0..W
13:39:01,582 value 0x156 is 1
13:39:01,582 Closing io device
```

```
p4 getv -l 15 -a 0x1c0
13:39:31,384 Opening '/dev/ttyUSB0' succeeded!
13:39:31,487 -> 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:39:31,499 got 02
13:39:31,499 got FD
13:39:31,499 got 00
13:39:31,499 got 0B
13:39:31,499 got 22
13:39:31,499 got 54
13:39:31,499 got 65
13:39:31,499 got 73
13:39:31,499 got 63
13:39:31,499 got 68
13:39:31,499 got 74
13:39:31,499 got 20
13:39:31,499 got 3B
13:39:31,499 got 2D
13:39:31,499 got 29
13:39:31,499 got 1C
13:39:31,499 <- 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:39:31,510 -> 02 FD 00 03 30 01 C0 17    ....0...
13:39:31,515 got 02
13:39:31,515 got FD
13:39:31,515 got 00
13:39:31,515 got 03
13:39:31,515 got 30
13:39:31,515 got 00
13:39:31,515 got 01
13:39:31,515 got 57
13:39:31,515 <- 02 FD 00 03 30 00 01 57    ....0..W
13:39:31,526 value 0x1c0 is 1
13:39:31,526 Closing io device
```

```
p4 getv -l 15 -a 0x1c9
13:40:10,078 Opening '/dev/ttyUSB0' succeeded!
13:40:10,181 -> 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:40:10,195 got 02
13:40:10,195 got FD
13:40:10,195 got 00
13:40:10,195 got 0B
13:40:10,195 got 22
13:40:10,195 got 54
13:40:10,195 got 65
13:40:10,195 got 73
13:40:10,195 got 63
13:40:10,195 got 68
13:40:10,195 got 74
13:40:10,195 got 20
13:40:10,195 got 3B
13:40:10,195 got 2D
13:40:10,195 got 29
13:40:10,196 got 1C
13:40:10,196 <- 02 FD 00 0B 22 54 65 73 63 68 74 20 3B 2D 29 1C    ...."Tescht ;-).
13:40:10,207 -> 02 FD 00 03 30 01 C9 0C    ....0...
13:40:10,211 got 02
13:40:10,211 got FD
13:40:10,211 got 00
13:40:10,211 got 03
13:40:10,211 got 30
13:40:10,211 got 00
13:40:10,211 got 03
13:40:10,211 got 51
13:40:10,211 <- 02 FD 00 03 30 00 03 51    ....0..Q
13:40:10,222 value 0x1c9 is 3
13:40:10,222 Closing io device
```

Kannst du damit etwas anfangen und was ist `...."Tescht ;-).`?

Danke und Gruß Hoppel

horchi, 2024-01-22:
okay da kommt wie man sieht kein Text zu dem Status mit sondern nur der Status als Zahl.
Bleibt nur die Übersetzung von Zahl in Text im p4d zu hinterlegen. Dazu brauche ich eine Liste der Status mit deren Bedeutung und noch eine Idee ob das dann für alle Kessel zutrifft.

AW0lke, 2024-02-11:
> okay da kommt wie man sieht kein Text zu dem Status mit sondern nur der Status als Zahl. Bleibt nur die Übersetzung von Zahl in Text im p4d zu hinterlegen. Dazu brauche ich eine Liste der Status mit deren Bedeutung und noch eine Idee ob das dann für alle Kessel zutrifft.

Ich stehe gerade auch vor der Herausforderung über MQTT die Betriebsart eines Heizkreises ändern zu wollen. 
@horchi, zu deiner Anmerkung, ob die Bedeutung für alle Kessel zutrifft: nein! Folgendes habe ich bei mir ausgelesen:

AUS: -2 (65534)
Auto: 0
Extraheizen: 1
Absenken: -1 (65535)
Dauerabsenken: -1 (65535)
Party: 1

hoppel118, 2024-02-15:
> Folgendes habe ich bei mir ausgelesen:
> 
> AUS: -2 (65534) Auto: 0 Extraheizen: 1 Absenken: -1 (65535) Dauerabsenken: -1 (65535) Party: 1

Darf ich Fragen, wie du das ausgelesen hast?

AW0lke, 2024-02-17:
> 
> Darf ich fragen wie du das ausgelesen hast?

Über das IO Setup der P4d sehe ich die Adresse bei Messwerte. 
Gleiches Ergebnis erhalte ich mit folgendem Befehl:
p4 getv -l 15 -a 0x0017