horchi/linux-p4d #48: "Pelletlager Restbestand" und "Pelletverbrauch-Zähler"

jkoopmann, 2024-01-08:
Hallo,

in der Fröling App sehe ich unter "Austragung" die Punkte
- Pelletlager Restbestand und
- Pelletverbrauch-Zähler

die ich wahnsinnig gerne in der p4d hätte. Leider finde ich die korrespondierenden Messwerte nicht. Gibt es die nicht oder bin ich einfach zu blind?

LG
   JP

MiGeppert, 2024-01-08:
Hallo JP,

den "Pelletlager Restbestand" finde ich ich meinem Setup auch nicht (*), der Pelletverbrauch-Zähler ("Pelletverbrauch Gesamt") ist bei meinem SP Dual auf "VA:0x114" (gefunden in der Weboberfläche).

(*)
Was ich interessant finde (und horchi vielleicht kommentieren könnte):
Unter "Service Menü" - "Austragung" - "Verbrauch" sehe ich blau unterlegt und klickbar "Pelletlager Restbestand: x.x t", beim draufklicken erscheint "Adresse 0x461, Typ 0x7".
Sowas finde ich aber leider nicht unter "Setup" - "IO Setup".

Warum ist das da nicht enthalten? 

Gruß
MiGeppert

jkoopmann, 2024-01-08:
Pelletverbrauch-Gesamt ist bei mir die Tonnenzahl seit Anbeginn der Zeit. Das ist besser als nichts aber für den "kurzfristigen" Verbrauch leider sehr ungenau.

Pelletverbrauch-Zähler in der App ist die Tonnenzahl (mit drei Nachkommastellen) seit dem letzten Reset. Das ist dann deutlich genauer. 

Aber am Ende suche ich in erster Linie "Pelletlager Restbestand" also den Wert, den Du gefunden hast.

horchi, 2024-01-09:
>Sowas finde ich aber leider nicht unter "Setup" - "IO Setup".
dort findest du nur die Messwerte, die Parameter der Service-Menü findest du nur dort - also im Service-Menü

MiGeppert, 2024-01-09:
Hallo horchi

> 'unter "Setup" - "IO Setup"' dort findest du nur die Messwerte, die Parameter der Service-Menü findest du nur dort - also im Service-Menü

Verstehe - aber der "zugehörige" Messwert wird bei mir an dieser Stelle angezeigt - hab ich ne Chance, dessen Adresse an dieser Stelle (oder sonstwo) herauszufinden?

Merci!

MiGeppert, 2024-01-09:
[image]

Die "2,9t" suchen wir :-)

horchi, 2024-01-10:
Verstehe ich das richtig das ist ein Parameter den man ändern kann welcher aber dennoch von der Heizung automatisch durch den aktuellen Verbrauch angepasst wird?

MiGeppert, 2024-01-10:
Korrekt!

Ich "tanke" x t Pellets und trage den Wert dann da ein (oder zähle die x t dazu).
Bei der Entnahme wird dann der Wert entsprechend verringert und ich weiß, wie der Stand im Pelletslager ist.

horchi, 2024-01-10:
zum Setzen und Abfragen der Parameter gibt es im Moment zum einen das Service Menü im WEBIF und zum anderen ein MQTT Interface welches [Hier](https://www.holzheizer-forum.de/forum/thread/50090-fr%C3%B6ling-announce-p4d-visualisierung-und-einstellung-der-s-3200-via-com1/?postID=218915#post218915 ) beschrieben ist:  (und Beiträge darunter).
Um sie genau wie die Messwerte auf dem Dashboard anzeigen zu können müsste ich den p4d erweitern, das aber dann  manuell konfigurierbar denn alle Parameter unter im IO Setup aufzulisten macht m.E. keinen Sinn - die meisten können ja nur manuell geändert werden.

horchi, 2024-01-10:
leider hat meine Firmware dieses Parameter noch nicht, könnte ich auch gut gebrauchen.
Dafür habe ich im p4d WEBIF die Pellets Seite hinzugefügt, da kann man eintragen was man getankt hat und bekommt eine Hochrechnung des Verbrauches über die Laufzeit der Stokerschnecke. Dazu muss man im setup "Pellet Verbrauch / Stoker Stunde" in kg einstellen, bei mir ~5,4 kg

MiGeppert, 2024-01-10:
Das macht die Steuerung selbst!
Ist nur die Frage, wie man an den Wert kommt!

jkoopmann, 2024-01-10:
> zum Setzen und Abfragen der Parameter gibt es im Moment zum einen das Service Menü im WEBIF und zum anderen ein MQTT Interface welches [...]

Ok. Da arbeite ich mich gerade rein, weil das auch mein anderes Problem lösen könnte, den Kessel ein- und auszuschalten über MQTT.

Für mich Blinden. :-)

mosquitto_pub -t p4d2mqtt/s3200/request -p 1883 -m { "command" : "parget", "address" : XX }

(bzw. das durch node_red schicken lassen). Das liest dann den Wert an der Adresse XX aus und das kommt dann über MQTT zurückgeflogen und ich kann es Parsen und in node_red verarbeiten? Wenn dem so ist (ich probiere das gleich aus): Wie komme ich an das passende XX heran? Im Service-Menü (per Web) sehe ich Adresse 0x461, Typ 0x7. Nehme ich als Adresse dann 0x461 oder 1121 ?

Dahinter steckt die Restzahl in Tonnen. Und wie es @MiGeppert beschreibt: Die S3200 zählt das eigenständig runter.

Hast Du eine Adresse/Parameter für mich, mit dem ich den Kessel ein-/ausschalten kann (gerne über MQTT!).

>(und Beiträge darunter). Um sie genau wie die Messwerte auf dem Dashboard anzeigen zu können müsste ich den p4d erweitern, das aber dann manuell konfigurierbar denn alle Parameter unter im IO Setup aufzulisten macht m.E. keinen Sinn - die meisten können ja nur manuell geändert werden.

Das wäre natürlich der Königsweg, weil man dann für den einen Parameter nicht periodisch abfragen muss. Geht aber in meinem Fall der node_red und nen Timer auch.

MiGeppert, 2024-01-10:
Das mit der Adresse muss mir horchi auch noch erklären, das Problem kann ich aber erst heute Abend von zu Hause aus genauer beschreiben.
Kessel (=Pelletseinheit?) kann man - soweit ich das mitbekommen habe, nicht über die Schnittstelle ein-/ausschalten...hätte ich auch gerne, da ich eigentlich mit Stückholz feuern mag und nur im Notfall (und von unterwegs) die Pelleteinheit aktivieren mag.
Vielleicht müsste da mal jemand beim Fröling Support anfragen!?

jkoopmann, 2024-01-10:
Mit 

`mosquitto_pub -t p4d2mqtt/s3200/request -p 1883 -m '{ "command" : "parget", "address" : 1121 }'`

passiert nichts.

Ich habe im Setup schon ergänzt:

[image]

Aber noch nicht neu gestartet. Wo bin ich falsch abgebogen?

horchi, 2024-01-10:
was kommt denn wenn du das machst im log des p4d ?

horchi, 2024-01-10:
> Das macht die Steuerung selbst! (also bei meiner Steuerung jedenfalls) Ist nur die Frage, wie man an den Wert kommt!

bei dir ja, wie gesagt meine hat diesen Parameter noch nicht (die ist von 2009)

jkoopmann, 2024-01-10:
Gar nichts. Als wenn er es nicht empfängt.

Sent from a mobile device.

horchi, 2024-01-10:
wenn das topic in Setup eingetragen ist muss er es empfangen. Ggf. noch den Log Level für MQTT aktivieren sofern der aus ist:
[image]

jkoopmann, 2024-01-10:
`Jan 10 18:09:39 raspberrypi p4d: <- (p4d2mqtt/s3200/request) [{ "command" : "parget", "address" : 1121 }] retained 0
Jan 10 18:09:39 raspberrypi p4d: Error: Ingnoring unexpected message in 'p4d2mqtt/s3200/request' (dispatchOther) [{ "command" : "parget", "address" : 1121 }]`

MiGeppert, 2024-01-10:
Hallo horchi

So, hier mal - ein wenig off-topic - meine Frage zu den Adressen.
Mit etwas Glück finde ich aber auf diesem Weg die Adresse zum Pellets-Restbestand.

Ich hab im WebIF (unter "Setup" - "IO Setup") zum Beispiel die "Kollektor Vorlauftemperatur" unter der Adresse "VA:0x0151", hier aktuell 21°C (wir im Heizraum gemessen, passt also).
Mit "p4 values -d /dev/ttyUSB_HEAT > p4_values.txt" finde ich in der TXT die Zeile
"65) 0x0151    2 '°' (0002) 'Kollektor Vorlauftemperatur'"
Stellt sich mir die Frage, was "2" und "(0002)" bedeutet, "0x0151" als Adresse und "'°'" als Einheit verstehe ich.
Mit "p4 getv -a 0x0151 -d /dev/ttyUSB_HEAT" bekomme ich aber
"value 0x151 is 42 / 42" - verstehe ich nicht! Warum "42" und warum "42 / 42"?

(Info: bei der Abgastemperatur unter Adresse 0x01 bekomme ich mit "p4 getv" den korrekten Temperaturwert angezeigt, aber auch wieder zweimal! aktuell "162 / 162", passt aber soweit)

Mit "p4 menu -d /dev/ttyUSB_HEAT > p4_menu.txt" finde ich in der TXT die Zeile
"307) Address: 0x0151, parent: 0x01f9, child: 0x0000; 'Kollektor Vorlauftemperatur'"
"Address" ist wieder klar, was bedeutet aber "parent" und "child"?

Mein Ziel ist übrigens, ein Skript zu schreiben, dass mir für alle Adressen aus "p4_values.txt" die Werte ausgibt und ich so vielleicht die "2,9t" zu finden.
Wenn ich haber hier nicht mal die 21°C Kollektorvorlauftemperatur finde, brauch ich eigentlich nicht weitermachen...

Für Ideen und Aufklärung wär ich dankbar :-)

horchi, 2024-01-10:
> `Jan 10 18:09:39 raspberrypi p4d: <- (p4d2mqtt/s3200/request) [{ "command" : "parget", "address" : 1121 }] retained 0 Jan 10 18:09:39 raspberrypi p4d: Error: Ingnoring unexpected message in 'p4d2mqtt/s3200/request' (dispatchOther) [{ "command" : "parget", "address" : 1121 }]`

Sieht für mich nach einer nicht ganz aktuellen Version aus?

jkoopmann, 2024-01-10:
Ja das vermute ich auch gerade und suche gerade die Anleitung, wie ich ein Update machen kann…. ☹ Verzeih die Blödheit. Du hättest nicht zufällig einen Link? :-=)

jkoopmann, 2024-01-10:
Kaum macht man es richtig....

`Jan 10 18:30:04 raspberrypi p4d: <- (p4d2mqtt/s3200/request) [{ "command" : "parget", "address" : 1121 }] retained 0
Jan 10 18:30:04 raspberrypi p4d: Perform MQTT command 'parget' for address 0x461
Jan 10 18:30:04 raspberrypi p4d: -> (p4d2mqtt/s3200/reply)[{"address": 1121, "unit": "t", "digits": 1, "min": 0.0, "max": 1000.0, "default": 500.0, "value": 3, "status": "success", "title": "Pelletlager Restbestand", "id": 205}]
Jan 10 18:30:04 raspberrypi p4d: Address: 0x0461; Unit: t; Value: 3.2 [{"address": 1121, "unit": "t", "digits": 1, "min": 0.0, "max": 1000.0, "default": 500.0, "value": 3, "status": "success", "title": "Pelletlager Restbestand", "id": 205}]`

jkoopmann, 2024-01-10:
Bleibt nur die Frage an den großen Meister, ob Du einen Weg kennst/hast, wie man den Kessel ein und ausschalten kann! Den Rest kriege ich jetzt abgefackelt! Super!!!!

horchi, 2024-01-10:
Sehe ich mir morgen an, da muss ich sicher noch was anpassen

horchi, 2024-01-11:
> @horchi : Oh und ein Nachfolgeproblem: Deine Routine liest zwar korrekt die 3.2t aus. Aber im JSON wird das zu
>     * digits:1 und
>     * value:3 
> statt value:3.2
> 
> Ich vermute, das kann ich nicht ändern sondern ist was im Parsen im code?

'parget' für Werte mit Nachkommastellen mit 0.9.57 behoben

horchi, 2024-01-11:
> Bleibt nur die Frage an den großen Meister, ob Du einen Weg kennst/hast, wie man den Kessel ein und ausschalten kann! Den Rest kriege ich jetzt abgefackelt! Super!!!!

ich kenne keinen

horchi, 2024-01-11:
> "65) 0x0151 2 '°' (0002) 'Kollektor Vorlauftemperatur'"
> Stellt sich mir die Frage, was "2" und "(0002)" bedeutet, "0x0151" als Adresse und "'°'" als Einheit verstehe ich.

2 ist der Faktor für den Wert - kommst so von der Steuerung. 0002 ist der von der Steuerung gelieferte Datentyp.
All das trifft nur soweit zu wie mein reengineering der Binären Daten richtig war/ist.

> "307) Address: 0x0151, parent: 0x01f9, child: 0x0000; 'Kollektor Vorlauftemperatur'"
> "Address" ist wieder klar, was bedeutet aber "parent" und "child"?
Das ist der Aufbau des Service Menüs, also die Verkettung der Menüpunkte untereinander, nicht jeder Menüpunkte ist auch ein Wert/Parameter, viele sind einfach die Menüebenen.
Hier war das reengineering schon sehr komplex und ich kann bis heute auch noch nicht alle Werte welche zu einem Menüpunkt geliefert werden interpretieren.

jkoopmann, 2024-01-11:
> > Bleibt nur die Frage an den großen Meister, ob Du einen Weg kennst/hast, wie man den Kessel ein und ausschalten kann! Den Rest kriege ich jetzt abgefackelt! Super!!!!
> 
> ich kenne keinen

Aber ich ggf. seit wenigen Minuten. Der Fröling Support ist wirklich super.

Ich vermute, Du nutzt Modbus über die serielle Schnittstelle. Hast Du die offizielle ModBus Beschreibung von Fröling? Es gibt dort digitale Eingang. Der Digitale Eingang mit der ID 10004 ist die Kesselfreigabe. Über diese sollte man laut Support den Kessel ein- und ausschalten können.

horchi, 2024-01-11:
ohh wow hätte nie gedacht das sie auf solche Anfragen reagieren, echt toll.
Wir verwenden nicht MODBUS, aber die digitalen Eingänge gibt es auch über dieses Interface. Müssen wir mal suchen

horchi, 2024-01-11:
ich nehme an es geht um diesen:

```
MariaDB [p4]> select * from valuefacts where name like '%Kesselfreigabe%';
+---------+------+------------+------------+---------+-------+--------+------+---------+--------+---------------------+----------------+----------+------+---------+--------+
| address | type | inssp      | updsp      | groupid | state | record | unit | options | factor | name                | title          | usrtitle | res1 | choices | rights |
+---------+------+------------+------------+---------+-------+--------+------+---------+--------+---------------------+----------------+----------+------+---------+--------+
|      31 | DI   | 1387191050 | 1641899265 |       1 | D     | NULL   |      |       0 |      1 | Kesselfreigabe_0x1f | Kesselfreigabe |          | NULL | NULL    |      0 |
+---------+------+------------+------------+---------+-------+--------+------+---------+--------+---------------------+----------------+----------+------+---------+--------+
1 row in set (0,001 sec)

```

MiGeppert, 2024-01-11:
Heißt: wenn man ID 10004 umschaltet, geht dann das Symbol unten rechts auf meinem Fröling Touch von grün auf rot bzw. umgekehrt?
Dann mag ich das gerne im WebIF haben :-)

Gruß und Dank für die Anfrage beim Fröling Support!

horchi, 2024-01-11:
kannst ja mal damit anfangen den digitalen Eingang anzeigen zu lassen und schauen ob das geht. Das man ihn auch schalten kann muss ich dann noch einbauen. 
Das ist etwas Arbeit

jkoopmann, 2024-01-11:
Und zwar innerhalb von 2 Stunden! Sehr kompetent und super freundlich. Und das ModBus Zeugs ist komplett dokumentiert und offiziell. Da dürfte eigentlich alles oder fast alles mit möglich sein und man könnte COM2 nutzen.

Theoretisch könnte man das Ganze dann auch physisch etwas entkoppeln. Modbus TCP Adapter dran und etwas wie p4d im Docker laufen lassen. Ich schwärme. 😊 Siehe hier: https://loxwiki.atlassian.net/wiki/spaces/LOX/pages/1704984631/Fr+ling+Pelletskessel+RS232+an+Loxone+Modbus+TCP

horchi, 2024-01-11:
ich werde es nicht auf MODBUS umbauen, zumindest Mittelfristig nicht. Der Aufwand die Kommunikation neu zu machen ist mir zu groß

jkoopmann, 2024-01-11:
Verstehe ich. Träumen darf ja erlaubt sein. 😊

jkoopmann, 2024-01-11:
> Heißt: wenn man ID 10004 umschaltet, geht dann das Symbol unten rechts auf meinem Fröling Touch von grün auf rot bzw. umgekehrt? Dann mag ich das gerne im WebIF haben :-)

Nein. Ich bin jetzt etwas weiter. Man muss unter 

"Systemmenü/Anlage/Anlagenart/Kesseltyp/(ausgewählter Kesseltyp)/ Freigabekontakt vorhanden"

selbigen auf ja setzen. Dann geht die Lambdatronic von einem Freigabekontakt aus. Der ist eigentlich physisch an einer Klemme, scheint aber eben auch mit dem Digitalen Eingang zu gehen. Wenn dieser geschlossen ist, ist die Freigabe erteilt und die normale Automatik läuft. Ist er offen, wird der Kessel nicht freigegeben, auch wenn die Automatik sagt "Heizen bitte".

Das hat aber keinen Einfluss auf das Icon fürchte ich (kann es gerade nicht testen). Und mitprotokollieren hilft leider nicht @horchi . Aber das macht ja so gesehen auch Sinn. Es wäre wirklich ganz phänomenal wenn Du das einbauen könntest, dass man den Eingang über Web-IF (und für mich noch wichtiger über MQTT) setzen kann.

MiGeppert, 2024-01-11:
@jkoopmann: das wäre dann aber nicht ganz das, was ich mag/brauche!

Wenn ich das richtig verstehe, sind das zwei "unterschiedliche" Schalter, d.h. man kann nicht am Touch oder im WebIF die Pelleteinheit ausschalten. Die Pelleteinheit läft dann wohl nur, wenn beide Schalter gleichzeitig "freigeben".
Wenn am Touch ausgeschaltet wird, kann ich via WebIF nicht einschalten, wenn im WebIF ausgeschaltet wird, kann ich am Touch nicht wieder einschalten. (Richtig?)

jkoopmann, 2024-01-11:
Mir ist nicht klar, was Du nun genau erreichen möchtest. Ich bezog mir nur auf den Kessel. Ich möchte im Sommer wenn die Solaranlage genug Dampf haben wird vermeiden, dass der Kessel anspringt. Das tut er nämlich leider, wenn er morgens feststellt, dass der Puffer unten zu leer ist und er ihn daher hochheizen möchte (was nur für die Heizung relevant wäre, aber nicht fürs Warmwasser und da man im Sommer die Heizung nicht braucht)…. Daher hat mir der Fröling Techniker empfohlen, den ab einem gewissen Zeitpunkt schlicht auszuschallten. Mit der oben beschriebenen Kombination kann ich das, weil ich einfach die Kesselfreigabe wegziehe…

Zugegeben: Das kann ich notfalls auch mit einem potentialfreien Kontakt und nem Shelly tun und ggf. ist das auch der einfachere Weg wenn ich das recht bedenke… 😊

MiGeppert, 2024-01-11:
Meine Absicht ist "ähnlich".
Ich hab einen Kombiofen Pellets+Stückholz.
Wir wollen überwiegend Stückholz feuern, d.h. der Kessel (die Pelletseinheit) ist standardmäßig ausgeschaltet, damit keine Pellets verheizt werden. Nur im "Notfall" (aus Faulheit oder was auch immer) werden Pellets verheizt.
Nur möchte ich, bei Bedarf und auch von unterwegs, die Pelletseinheit aktivieren können (falls niemand zu Hause ist und die Bude bzw. der Pufferspeicher kalt ist. Aber auch andere Familienmitglieder sollen die Möglichkeit haben, die Einheit am Display zu aktivieren (weil gerade kein Handy/Notebook/usw. zur Hand).
Im Sommer übernimmt bei mir auch die Solarthermie (bei abgeschaltetet Pelletseinheit) und wenn's zu kalt ist, wird angefeuert (oder eben die Pelletseinheit aktiviert), damit ich Warmwasser hab, wenn ich heimkomme.