horchi/linux-p4d #50: S4 support

JamesPatageule, 2024-06-06:
Hello,

Is it plan to make P4D compatible with  ? So estimate/calculate for kg of logs of wood instead of pellet ?

Anyway, thanks for the great software !!!

horchi, 2025-02-24:
At the moment the p4d is compatible with all heaters that use the S3200 control unit, an extension is not planned for the time being

closed 2025-02-24 by horchi

JamesPatageule, 2025-02-24:
I have a S3200, but using wood and not pelets

horchi, 2025-02-24:
that should work without a problem

JamesPatageule, 2025-02-25:
The application works, but detect my S4 with Pellet ... 
My ticket was to know if we could correct this model detection or to have a setting somewhere maybe ?

And then if we could have a consummable following + stats for wood (like it seems to be for pellets).