horchi/linux-p4d #51: Flut von json Fehlern

SaimenSays, 2024-06-20:
Ich nutze die 'Installation by source' Methode, weil  Buster schon lange abgelaufen ist und kein neueres Paket zur verfügung steht. Aber irgendetwas läuft schief. Zum einen wird keine Logdatei /var/log/p4d.log angelegt.
Mit `service p4d status` sehe ich eine Flut von folgenden Fehlermeldungen. Kann mir jemand sagen woran das liegt und wie man es beheben kann?
```
Jun 20 22:03:07 p4d p4d[3605]: Error: Ignoring invalid json in ''
Jun 20 22:03:07 p4d p4d[3605]: Error decoding json: '[' or '{' expected near end of file (<string>, line 1 column 0, position 0)
Jun 20 22:03:11 p4d p4d[3605]: Error: Ignoring invalid json in ''
Jun 20 22:03:11 p4d p4d[3605]: Error decoding json: '[' or '{' expected near end of file (<string>, line 1 column 0, position 0)
Jun 20 22:03:22 p4d p4d[3605]: Error: Ignoring invalid json in ''
Jun 20 22:03:22 p4d p4d[3605]: Error decoding json: '[' or '{' expected near end of file (<string>, line 1 column 0, position 0)
Jun 20 22:03:22 p4d p4d[3605]: Error: Ignoring invalid json in ''
Jun 20 22:03:22 p4d p4d[3605]: Error decoding json: '[' or '{' expected near end of file (<string>, line 1 column 0, position 0)
Jun 20 22:03:26 p4d p4d[3605]: Error: Ignoring invalid json in ''
Jun 20 22:03:26 p4d p4d[3605]: Error decoding json: '[' or '{' expected near end of file (<string>, line 1 column 0, position 0)
```

horchi, 2024-06-21:
Von den Meldungen abgesehen funktioniert es?

Mach mal alle 'Log' Schalter an und pate es dann nochmal mit mehr Kontext.
Versuch auch mal ob es auch passiert wenn du alles in /etc/p4d/scripts.d/ löschst.

SaimenSays, 2024-06-21:
Danke für die schnelle Antwort!
Die Weboberfläche funktioniert unregelmäßig. Man muss öfters zwei oder dreimal klicken bis sich etwas tut. Ich konnte unter "Commands" die "Init Sensoren" auslösen. Unter "Liste" sehe ich die aktuellen Werte. Mqtt habe ich noch nicht ans laufen bekommen. Funktioniert das nur mit aktiviertem "Aufzeichen"?
Die Statusanzeige rechts oben blinkt grün.

In "Allg. Konfiguration" bei "Log Eloquence" alle Schalter aktiviert. Dienst neu gestartet, aber weiterhin kein Logfile.

Den Inhalt des Ornders habe ich gelöscht. Ausgabe hat sich verändert, aber nicht wirklich besser:
```
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
Jun 21 14:45:55 p4d p4d[4111]: loop ...
```

horchi, 2024-06-21:
die Log Meldung zeigt das der p4d von einem Topic eine leer Nachricht erhält aber JSON erwartet. 
Ich hatte gehofft das es von einem der Skripte kommt. Merkwürdig das es mit den erhöhten Log Level keine Meldung gibt woher die kommt.

Mach mal mit mosquito_sub -v '#' auf welches Topic diese Nachrichten  geschrieben werden. Muss ja in hoher Frequenz sein.

SaimenSays, 2024-06-22:
Mein MQTT Broker läuft auf einem anderen Server. Dort alle Topics abzurufen würde wohl in >1000 enden und wäre nicht sinnvoll es hier zu posten. Ich habe kein mosquitto auf dem p4d Rechner installiert. Ist dies das Problem?
Zum Test habe ich die Adresse aus "MQTT Broker Url" komplett entfernt. Die "loop ..." Meldungen werden aber weiterhin ausgegeben.

horchi, 2025-02-24:
es geht nur um die Topics welche der p4d liest. Für das langsame reagieren der Oberfläche gibt es nun dem Log Schalter LoopTimings. 
Ich mache das mal zu, wenn die Probleme noch bestehen bitte mit den Details einen neuen Issue aufmachen

closed 2025-02-24 by horchi